openmv/openmv #1626: sensor.set_pixformat(sensor.RGB565) error on OpenMV Cam H7

airsalutanonymous, 2022-04-24:
Hello, 

I am trying to set my camera sensor to RGB, but I can't.
I have read a lot on this issue, but it seems that my camera support RGB: https://www.sparkfun.com/products/retired/15325
This is the model I have. 

Thanks, 
Hugo 

Error message:
ValueError: Sensor control Failed!

```
import sensor, image, time

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.set_auto_whitebal(False)
sensor.skip_frames(time = 2000)

clock = time.clock()

while(True):
    clock.tick()
    img = sensor.snapshot()
    print(clock.fps())

```

kwagyeman, 2022-04-24:
Is the camera mounted correct?

airsalutanonymous, 2022-04-24:
What do you mean by mounted correct? 

[image]

Back of my sensor: 
[image]

kwagyeman, 2022-04-24:
You have the global shutter sensor installed which is grayscale only.

closed 2022-04-24 by kwagyeman