openmv/openmv #1631: Is there any plan to make OPENMV support ESP32-S3?

Micky513673326, 2022-05-06:
I have an ESP32-S3 based board. Equipped with camera and 4g mobile module. 
Too bad OPENMV doesn't currently support it. 
Does the maintainer have any plans to port it for support?

kwagyeman, 2022-05-06:
Yeah, it's really a matter of us having a financial reason to do. It's a lot of work.

Micky513673326, 2022-05-07:
I want to port this to my ESP32S3. What documents do I need to refer to to get started?

iabdalkader, 2022-05-08:
We don't have any porting or development guides.

kwagyeman, 2024-02-16:
This will get done if a sponsor appears to pay for the work. Closing for now.

closed 2024-02-16 by kwagyeman

leeebo, 2024-10-23:
Porting openmv to [ESP32-P4](https://www.espressif.com.cn/en/products/socs/esp32-p4) (400MHz Dual Core RISC-V) would make more sense, Espressif has already ported the [opencv ](https://components.espressif.com/components/espressif/opencv/versions/4.10.0~2) on it.

kwagyeman, 2024-10-23:
Our next-gen product will out perform that system by 300x.

ESP32 missed the target with the P4.