openmv/openmv #1632: Unable to build firmaware

tejalbarnwal, 2022-05-10:
**Describe the bug**
I have recently forked the original repository to build the firmware to run the model on the NICLA Vision sensor. The firmware build fails at the development release step.

**To Reproduce**
I have replaced the files in the `openmv/src/lib/libtf/models/` by my [`.bin`](https://github.com/tejalbarnwal/openmv/tree/master/src/lib/libtf/models) and [`.txt`](https://github.com/tejalbarnwal/openmv/tree/master/src/lib/libtf/models) file obtained from Edge Impulse and OpenMV dataset respectively. Upon the activating the github action, I receive an error stating 
`🌶  failed to delete ref "refs/tags/development" <- Resource not accessible by integration`
`🌶  An error occured while deleting the tag "development"`

**Screenshots**
[image]

**Please complete the following information**
 - IDE version: 2.9.0
 - Firmware vesion: latest, directly forked `master` (https://github.com/tejalbarnwal/openmv)
 - Board: Arduino Nicla Vision
 - OS: Windows

iabdalkader, 2022-05-10:
Try creating the `development` tag, i.e:

```bash
git tag development
git push origin development
```

tejalbarnwal, 2022-05-11:
The tag `development` already exists. Also, does this push has to go to the `development` branch.? To deploy the model from the edge impulse, I went into the deployment section in the menu of edge impulse, selected OpenMV under "Build firmware" and clicked "build" with quantized optimization as stated [here in the Deploy section](https://docs.arduino.cc/tutorials/nicla-vision/image-classification). The `.zip` file that gets downloaded returns `.bin` files and not `.tflite`.

iabdalkader, 2022-05-11:
I think that `bin` you have is a firmware image with the model embedded.

This, what you're trying to do, is something else entirely. This is to add your own built-in model and build the firmware from source (the easy way).

Note your screenshot doesn't show the errors, just that something fails, I'm not sure if I'm missing something.

> Also, does this push has to go to the development branch

There's no development branch, and the tag isn't needed either (I forgot that we don't need that anymore). I'll fork the repo and see what's wrong.

tejalbarnwal, 2022-05-11:
Yes, right! I meant the `delete old release` step is giving the same original error. I am not sure but could it be an issue related to `GITHUB_TOKEN`?
https://docs.github.com/en/actions/security-guides/automatic-token-authentication#permissions-for-the-github_token indicates for forked repos the `GITHUB_TOKEN` only has ‘read’ permission

iabdalkader, 2022-05-11:
> . I am not sure but could it be an issue related to GITHUB_TOKEN?

No, it's because the development release doesn't exist, so that step fails and the workflow fails. I'm trying to fix it.

iabdalkader, 2022-05-11:
I think you can fix it in your fork, but I also want to fix it in the workflow.

[image]

tejalbarnwal, 2022-05-11:
oh, could you please let me know when you correct the workflow too?

closed 2022-05-11 by iabdalkader

tejalbarnwal, 2022-05-11:
Yes, it is working perfectly! Thank you so much!😊

PR #1633 (2022-05-11): github: Fix firmware workflow permissions.
Fixes #1632