openmv/openmv #1635: FEATURE ADD: IDE WILL SEND SYSTEM TIME SOON

kwagyeman, 2022-05-14:
On detection of firmware 4.3.2 or above the IDE will now start to send the system time to the OpenMV Cam. The IDE will send the following on connect:

1st USB Packet:

1 byte == __USBDBG_CMD
1 byte == __USBDBG_TIME_INPUT (0x12)
4 bytes == 32 (for a 32-byte struct)

2nd USB Packet:

4-bytes (year)
4-bytes (month)
4-bytes (day)
4-bytes (dayOfWeek)
4-bytes (hour)
4-bytes (minute)
4-bytes (second)
4-bytes (msec)

...

This feature must be implemented in the firmware before version 4.3.2 ships.

iabdalkader, 2022-05-15:
I've added placeholders for the new commands, feel free to implement.

closed 2022-05-15 by iabdalkader

iabdalkader, 2022-05-15:
Note the next release is 4.3.1 not 4.3.2. The value in usbdbg.h is the next release not the current one.

kwagyeman, 2022-05-16:
I specifically put it one release ahead.

PR #1641 (2022-05-15): ports/all: Add placeholders for new USBDBG commands.
* Fixes  #1634.
* Fixes  #1635.
commit 42bd64d8 (2022-05-15): ports/all: Add placeholders for new USBDBG commands.

* Fixes  #1634.
* Fixes  #1635.