openmv/openmv #1646: fast float rounding APIs in fmath.h can be optimized to use VRINT family instructions

RockySong, 2022-05-25:
openmv uses VCVT family instructions, they rely on FPSCR[23:22]=0 (default settings) for rounding policy, and in fast_ceilf():

int OMV_ATTR_ALWAYS_INLINE fast_ceilf(float x)
{
    int i;
    x += 0.9999f;
    asm volatile (
            "vcvt.S32.f32  %[r], %[x]\n"
            : [r] "=t" (i)
            : [x] "t"  (x));
    return i;
}

"0.9999f" is not enough to get max float number that < 1.0f (0x3F7FFFFF)

Suggest to use "VRINTP.S32" in fast_ceilf, and "VRINTM.S32" in fast_floorf, VRINTA/VRINTN for normal rounding.

iabdalkader, 2022-06-08:
Thanks @RockySong for reporting this. I fixed `ceilf` and `floor` as suggested in https://github.com/openmv/openmv/pull/1663 but I'm thinking I should leave `roundf` as is (using `vcvtr`) because it allows overriding the rounding mode by writing `FPSCR` (if needed), possibly by defining something in a board configuration or from Python code (by writing to that register if possible).

RockySong, 2022-06-11:
That's reasonable and you are welcome!

PR #1663 (2022-06-09): imlib: Optimize fmath ceilf and floorf functions.
Fixes #1646