openmv/openmv #1696: [bug] LCD shield y_scale=-1 not filp

kidswong999, 2022-07-22:
code:
```
import sensor, image, lcd
sensor.reset() # Initialize the camera sensor.
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QQVGA2)
lcd.init() # Initialize the lcd screen.

while(True):
    lcd.display(sensor.snapshot(), y_scale=-1) 
```

The document says that if the y_scale is negative, it will be flipped, but the LCD shield is not flipped.

x_scale=-1 works fine.

iabdalkader, 2022-09-22:
@kwagyeman Is this related to the scaling bug ?

kwagyeman, 2022-09-22:
No, this was ROI related.

iabdalkader, 2022-09-22:
I mean is it fixed by the recently merged patch ?

kwagyeman, 2022-09-22:
Unsure, probably not. I can add this to fix right now.

kwagyeman, 2022-09-26:
Hi, this can't be fixed without turning triple buffering on.

Draw image works by processing the image from the top left to bottom right. When you specify the image to be flipped it just outputs the pixels that it's processing from bottom to top versus top to bottom. However, when the LCD shield is used without triple buffering there's no display buffer so the bottom to top lines are just sent to the LCD immediately which just displays them top to bottom.

I'll add a error note if you try to do this without triple buffering on.

(Making draw image process the image top to bottom when y_scaled is -1 would require writing the whole piece of code).

kwagyeman, 2022-09-26:
https://github.com/openmv/openmv/issues/1696

closed 2022-09-26 by iabdalkader