openmv/openmv #1697: Update firmware with latest version of ulab

tejalbarnwal, 2022-07-25:
[Here](https://forums.openmv.io/t/ulab-numpy-dimensions/7339) is the link to the issue on the OpenMV forum that I was facing concerning the implementation of more than two dimensions of NumPy arrays.
The author for `ulab` just stated that this bug was fixed recently. This could be found [here](https://github.com/v923z/micropython-ulab/issues/498#issuecomment-1194283767)

It would be really helpful if the parameter for `ULAB_MAX_DIMS` could be set to 4 instead of 2 to be used on OpenMV products. 

**Please complete the following information**
 - IDE version: [2.9.2]
 - Firmware version: [4.3.1]
 - Board: [Arduino Nicla Vision]
 - OS: [e.g. Windows]

iabdalkader, 2022-07-25:
Don't you mean 3 instead of 2 ? Why do you need a 4D array ?

iabdalkader, 2022-07-25:
This adds about 20KBs to the firmware image, so there must be a very good reason to enable 3D arrays.

tejalbarnwal, 2022-07-26:
I needed 3D arrays to convert RGB images to NumPy arrays

tejalbarnwal, 2022-07-27:
Hey, new functionality was added to the ulab as found [here](https://github.com/v923z/micropython-ulab/issues/527). Could these be added to the firmware of the board that I am using. How can I do so?

iabdalkader, 2022-07-27:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/ulab-numpy-dimensions/7339/4

tejalbarnwal, 2022-07-27:
Hey, I have a very dumb doubt, updating the firmware with modifications corresponding to `ulab` would allow me to use new functions in OpenMV IDE too, right?

iabdalkader, 2022-07-27:
Yes.

tejalbarnwal, 2022-07-27:
Great, thank you so much!
Am sorry to write it here, I will edit this comment later but this is the only medium I can talk to you. Could you please https://forums.openmv.io/t/find-displacement-with-usb-vcp/7349 at this issue too? I will be really grateful to you for this!

tejalbarnwal, 2022-07-29:
> To update ulab you'll have to update the submodule here omv/modules/ulab, google how to do that or just replace it with a newer ulab and rebuild.
Hey, I followed the following steps to update the submodule
```bash
cd omv/modules/ulab
# this is the branch I want 
git checkout nonzero
# Update
git pull
```
And then committed to changes to the forked repository, but there seems to error in building the firmware for all the boards except OPENMV2
[image]

All the boards except OPENMV2 give the same error regarding ` undefined reference to io_load_obj, io_loadtxt_obj, io_save_obj, io_savetxt_obj `. @iabdalkader Any idea what am I missing?

tejalbarnwal, 2022-07-29:
Hey, I solved it
Thanks to @v923z

iabdalkader, 2022-07-29:
Good, I'll update ulab soon.

iabdalkader, 2022-09-18:
@v923z Any plans for a new stable release soon or a tag after https://github.com/v923z/micropython-ulab/commit/00139bdd9684bd9d10b0d89fa8fbb0906f2d05af please ?

v923z, 2022-09-19:
@iabdalkader There is a pending issue that might possibly affect you, too. As soon as that is fixed (should happen tomorrow, or the day after tomorrow), I'll issue a new tag.

PR #1731 (2022-09-26): micropython: Update ulab to 5.1.1.
* Fixes #1697.
commit 2ebcfc60 (2022-09-26): micropython: Update ulab to 5.1.1.

* Fixes #1697.
commit da36b9df (2022-09-26): micropython: Update ulab to 5.1.1.

* Fixes #1697.
closed 2022-09-26 by iabdalkader