openmv/openmv #1739: Feature Request: Add support for linear data ML models

tejalbarnwal, 2022-10-07:
The issue is related to the [this](https://forums.openmv.io/t/running-a-non-linear-regression-model-on-processed-data/7550/9) discussion on the forum.

iabdalkader, 2022-10-07:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/running-a-non-linear-regression-model-on-processed-data/7550/10

tejalbarnwal, 2022-10-14:
Hey, I am not able to find the `libtf_builtin_models.h` file neither in this repository or `openmv/tensorflow-lib` repo. Could you please help me?

iabdalkader, 2022-10-16:
> find the `libtf_builtin_models.h` file

This file is auto-generated, also my reply above is automated.

tejalbarnwal, 2022-10-17:
Hey @iabdalkader @kwagyeman , 
I implemented the function myself in the `py_tf.c`, but now I want to debug it. Is there any way to debug the code without running it on OpenMV IDE.
The code I am running right now doesn't involve any input from the development board I am using.
```python
import time, os, tf

labels, net = tf.load_builtin_model('force_quant')

print("input height:", tf.input_height(net))
print("input width:", tf.input_width(net))
print("input channels:", tf.input_channels(net))
print("input datatype:", tf.input_datatype(net))
print("input scale:", tf.input_scale(net))
print("input zero point:", tf.input_zero_point(net))

print("output height", tf.output_height(net))
print("ouput width", tf.output_width(net))
print("ouput channels", tf.output_channels(net))
print("ouput datatype", tf.output_datatype(net))
print("op scale:", tf.output_scale(net))
print("op zero point:", tf.output_zero_point(net))

input_list = [-8, -5, -8,  2, -7, 10, -4, 12, -4, -2, -4,  7, -3, 14,  1, -6,  1, 3,  2, 11,  2, 13,  6, -2,  8,  7,  7, 13,  7, -5, 10,  2, 11,  9, 9, 11]

output = tf.regression(net, 1, input_list)
print(output)
```
I just wanna debug the `tf.regression` function that I implemented. I am currently using `printf` statements and building the firmware and deploying it each time.
Is there any easier way to debug it that you know? maybe running the whole thing on linux machine instead of deploying the firmware everytime on the board?

iabdalkader, 2022-10-18:
Yes, you can use jlink to debug (build the firmware with `DEBUG=1`).

tejalbarnwal, 2022-10-18:
Just to confirm the process, i will use [this](https://github.com/openmv/openmv/blob/master/src/README.md#build-artifacts) link to follow the commands which build the openmv repo which a modification of following commands with `DEBUG =1`
```bash
make -j$(nproc) -C src/micropython/mpy-cross DEBUG=1 # Builds Micropython mpy cross-compiler 
make -j$(nproc) TARGET=<TRAGET_NAME> -C src DEBUG=1 # Builds the OpenMV firmware
```
And then i do
``` bash
cd ~/openmv/src/micropython/ports/unix
./micropython
```
This would allow me to import all the modules built for openmv, right?

iabdalkader, 2022-10-22:
I really have no idea, maybe the format is not correct, how did you train that model ? Should try EdgeImpulse see if it works.

kwagyeman, 2022-10-23:
So busy. I will get to it eventually.

tejalbarnwal, 2022-10-23:
Hey, I solved the problem it the day before yesterday. Now I could use any regression model that takes 1-D input. Can I make a PR it after some days after I am done with my research project?

iabdalkader, 2022-10-23:
@tejalbarnwal Yes of course, you're more than welcome. But please follow the guidelines when sending the PR, it should make reviewing the PR much faster:

https://github.com/openmv/openmv#contribution-guidelines

tejalbarnwal, 2022-12-26:
@iabdalkader I have submitted a PR mentioned above in `tensorflow-lib` repo for the support for 1D regression models

closed 2022-12-28 by iabdalkader

tejalbarnwal, 2023-06-01:
Hey @iabdalkader , I updated the functionality to work with 2D arrays-based ulab module as well. Shall I open up a PR for the same?

iabdalkader, 2023-06-01:
> Shall I open up a PR for the same?

Yes go ahead, thank you!

PR #1763 (2022-12-28): lib/libtf: Add support for 1D/1D regression models.
* Fixes #1751.
* Fixes #1739.
commit ac49aa52 (2022-12-28): lib/libtf: Add support for 1D/1D regression models.

* Fixes #1751.
* Fixes #1739.