openmv/openmv #1741: Can't load person_detection model in firmware 4.3.3

kidswong999, 2022-10-13:
Hi, Is person_detection removed in new firmware?

Error Code:
```
import tf
labels, net = tf.load_builtin_model('person_detection')
labels, net = tf.load('person_detection')
```

tejalbarnwal, 2022-10-13:
Hey, I think the source code for firmware 4.3.3 has `fomo_face_detection` as a built-in model.
 In order to use person detection model you would have to replace the model in [this](https://github.com/openmv/openmv/tree/master/src/lib/libtf/models) directory with `person_detection.tflite` and `person_detection.txt` and build the firmware again to use that if you would want to stick to features of 4.3.3

iabdalkader, 2022-10-13:
Hi yes, it takes more than 400KBs of flash, removed it to save space and now the default demo is fomo face detection.

closed 2022-10-13 by iabdalkader