openmv/openmv #1750: OpenMV H7 Cam does not stream video via RTSP

vsarun56, 2022-11-24:
While using rtsp_video_server_wlan1.py from example in the OpenMV IDE. I try to stream video with vlc, but I don’t get any video streaming.

play_callback(pathname, session):
print (“Playing "%s" in session %d” % (pathname, session))

in the openmv ide terminal, I only see play_callback output.
[image]

kwagyeman, 2022-11-24:
Hi, I need to debug the script. It's on my todos.

kwagyeman, 2023-02-15:
@iabdalkader - I found the issue. The WiFi code is broken in regards to being able to accept socket connections:

```
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.bind(self.__myaddr)
print(s)
s.listen(0)
print(s)
s.settimeout(5)
print(s)
self.__tcp__socket, self.__client_addr = s.accept()
print(s)
s.close()
print(s)
```

The code will Just print:

<socket state=0 timeout=-1 incoming=0 off=0>
<socket state=1 timeout=-1 incoming=0 off=0>
<socket state=1 timeout=5000 incoming=0 off=0>

And then hang on accept. It should timeout. Also, if a connection tried to be made the camera will just not respond to the remote device at all resulting in a TCP re-transmission.

[image]

iabdalkader, 2023-02-16:
@kwagyeman How do I reproduce this ? I tried the following and it times out as expected:

```python
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.bind(["", 8000])
print(s)
s.listen(0)
print(s)
s.settimeout(5)
print(s)
client = s.accept()
print(s, client)
```

kwagyeman, 2023-02-16:
Run the RTSP example and try to connect to it via FFMPEG. You just need to pass do "ffplay rtsp://xxx.xxx.xxx.xxx".

What I observed in wire shark is that the accept never returns and on wire shark I see the PC sends the SYN packet to setup the TCP socket and gets no response.

kwagyeman, 2023-02-20:
This is fixed in: https://github.com/openmv/openmv/pull/1786

closed 2023-03-08 by iabdalkader