openmv/openmv #1751: Making Predictions on arrays with tflite model

BolajiLawal, 2022-11-24:
**Describe the bug**
I am currently sending data from a wireless sensor to the openmv using uart. The idea is to use a pretrained tflite model to make predictions on the data from the sensor which will be stored as an array. The idea is to use the openmv to take a snapshot if the output from the prediction meets a certain condition/threshold. However, there is currently no way to use tflite model to make predictions on array

**Please complete the following information**
 - IDE version: 2.9.7
 - Firmware version: 4.3.2
 - Board: OPENMV-H7+
 - OS: Windows

chitraa, 2022-12-09:
Hey since I have a similar requirement, I am open adding this support to openmv and raise a PR. 
Would need some direction @iabdalkader

closed 2022-12-28 by iabdalkader

BolajiLawal, 2023-03-12:
@chitraa @iabdalkader can you please explain to me how the tf.regression works. Is it possible to set the precision of the output i.e. if I want to 4 decimal places or something. If not what's the default precision, if I get an output of 0.998 will this be approximated to 1?

kwagyeman, 2023-03-12:
It's a floating point number. The precision is speced by the print statement.

BolajiLawal, 2023-03-25:
Thanks, so i have a tflite neural network model where the output layer is softmax i.e. should give an array of probabilities. My feeling is if have an input 1D array and use tf.regression, the output 1D array should be an array indicating the probability of the input belonging to each class in the model. But when I try with two different inputs, I get the same answer an outpur array of [0,1]. This is wrong because the examples belong to different classes so should give a different answer. Is there a way I can use tf.regression for this problem or am I doing something wrong?

PR #1763 (2022-12-28): lib/libtf: Add support for 1D/1D regression models.
* Fixes #1751.
* Fixes #1739.
commit ac49aa52 (2022-12-28): lib/libtf: Add support for 1D/1D regression models.

* Fixes #1751.
* Fixes #1739.