openmv/openmv #1776: MT9V0XX sensor clock frequency to high on Portenta H7

j-wagner-md, 2023-02-06:
On the Portenta H7 board the configured sensor clock frequency for the MT9V034 global shutter module (**26.6 Mhz**) is not working properly.  
Measuring it externally with an oscilloscope we see **close to 28.6 MHz**
[image]

The datasheet for the MT9V034 global shutter sensor specifies a maximum input clock frequency of 27.0 MHz.
[image]

**So the sensor will be used outside its specified limits.**

Attempts to change the sensor frequency to 27MHz or 26MHz were not successful and the sensor still showed the 28.6MHz system clock frequency (`src/omv/sensors/mt9v0xx.h` --> `MT9V0XX_XCLK_FREQ`).
Despite the Portenta having two oscillators on board. One with 25Mhz output and one with 27MHz.
[image]

Only setting it to **25MHz** showed the desired results of lowering the frequency into the specified frequency range (It measured 25MHz as expected)

**Setup information**
 - Firmware version: 4.4.1
 - Board: Portenta H7 Lite
 - Sensor: MT9V034 - OpenMV global shutter module

---
As a note:

Mounting the same sensor onto an OpenMV Cam H7 Plus with the same firmware hash (set with 26.6 MHz before the test changes) resulted in exact Clock speeds:
[image]

iabdalkader, 2023-02-06:
The Portenta has different OSC frequency and different PLL settings, it's probably not possible to get an exact 26.6MHz or 27MHz. As far as I know there aren't any working kits out there with the MT and the Portenta, that's why it hasn't been a priority to test/fix this, but maybe that has changed. I'll send a fix later to override the frequency for Portenta to 25MHz to keep it on the safe side.

closed 2023-02-06 by iabdalkader

j-wagner-md, 2023-02-07:
Thank you @iabdalkader for your fast support :+1:

PR #1778 (2023-02-06): boards/PORTENTA: Override default MT sensor clock.
Fixes #1776