openmv/openmv #1794: get_threshold function result is incorrect

kidswong999, 2023-03-02:
a_value and b_value result is incorrect .

```
import sensor, image, time

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time = 2000)

clock = time.clock()

while(True):
    img = sensor.snapshot()
    histogram = img.get_histogram()
    THRESHOLD = histogram.get_threshold()
    l  = THRESHOLD.l_value()
    a = THRESHOLD.a_value()
    b = THRESHOLD.b_value()
    print(THRESHOLD)
    img.binary([(0, l), (-128, a),(-128, b)])
```

**Expected behavior**
IDE show binary image with optimal thread. 

**Screenshots**
[image]

**Please complete the following information**
 - IDE version: 2.9.7
 - Firmware version: 4.3.3
 - Board: OpenMV4 Plus
 - OS: Windows

kwagyeman, 2023-03-02:
What output were you expecting? The method just finds a nice split value to threshold the image.

kwagyeman, 2024-01-03:
Hi, this is because you are passing the arguments to binary incorrectly. It's a list of 6 value tuples. Or a list of 2 value tuples.

```
import sensor, image, time

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time = 2000)

clock = time.clock()

while(True):
    img = sensor.snapshot()
    histogram = img.get_histogram()
    THRESHOLD = histogram.get_threshold()
    l  = THRESHOLD.l_value()
    a = THRESHOLD.a_value()
    b = THRESHOLD.b_value()
    print(THRESHOLD)
    img.binary([(0, l, -128, a, -128, b)])
```

However, I also found an issue with get_threshold. A/B arguments were not shifted. 

https://github.com/openmv/openmv/pull/2074

closed 2024-01-07 by iabdalkader