openmv/openmv #1811: bug with to_grayscale(copy = True)

jcmag, 2023-03-28:
**Describe the bug**
I want to detect white lines in my camera’s images, using this code:

img = sensor.snapshot()
blackImage = img.to_grayscale(copy = True).binary([(190, 255)])
segments = blackImage.find_line_segments(merge_distance = 10, max_theta_diff = 5)    
for segment in segments:
    img.draw_line(segment.line(), color = (255, 0, 0))

I’m calling to_grayscale() and binary() to detect the lines in a black and white image, to be more accurate.
But the last call (img.draw_line) doesn’t draw the lines in the displayed image (in the IDE).

If I put to_grayscale(copy = False), I see grey lines in the displayed image (the grey one).

And same issue with:
black_img = img.to_grayscale(copy = True)
black_img.binary([(190, 255)])

**Expected behavior**

I expect the lines to be drawn and displayed in the colored image.

kwagyeman, 2023-04-02:
Found the bug.  Very simple issue.

kwagyeman, 2023-04-02:
https://github.com/openmv/openmv/pull/1813

closed 2023-04-07 by iabdalkader