openmv/openmv #1819: Create a Flush Function for mjpeg for power safety

Tkdefender88, 2023-04-07:
**Describe the bug**
When creating a mjpeg file, if there is ever an error or a power failure before the board has the chance to call mjpeg.close() the file is corrupted and cannot be read from. Currently there is no way to periodically save contents of the video stream to the file to mitigate losing all the data in the event of a power failure.

To fix this add a flush function to mjpeg to save the current video stream to the sd card while keeping the file open so one can continue to add more frames.

See this discussion: https://forums.openmv.io/t/power-safety-with-saving-mjpeg-to-sd-card/8203

**To Reproduce**
Steps to reproduce the behavior:
1. create an mjpeg file and add frames to it. 
2. either power down the camera or neglect to call mjpeg.close
3. notice that the file exists on the SD card but is corrupted and cannot be viewed. The data lost.

**Expected behavior**
Add a flush function so that one can periodically call flush to save data to the file.

**Screenshots**
N/A

**Please complete the following information**
 - IDE version: 2.9.7
 - Firmware version: 4.4.1
 - Board: OPENMV-H7+
 - OS: Linux

iabdalkader, 2023-04-07:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/power-safety-with-saving-mjpeg-to-sd-card/8203/4

Tkdefender88, 2023-04-07:
I linked to that discussion in my ticket. I am the author of that post and kwagyeman told me to make a bug ticket here for the problem.

iabdalkader, 2023-04-08:
It's a bot ignore it.

kwagyeman, 2023-04-08:
https://github.com/openmv/openmv/issues/1819

closed 2023-04-15 by kwagyeman