openmv/openmv #1823: Compile opening source code error

Baidu-NJZY, 2023-04-15:


Baidu-NJZY, 2023-04-15:
error: initializer element is not constant
 #define ULAB_REFERENCE_FLOAT_CONST(id) ((mp_obj_t)(id))
                                        ^
/mnt/hgfs6d_16/openmv/openmv­c/omv/modules/ulab/code/numpy/numpy.c:84:35: note: in expansion of macro 'ULAB_REFERENCE_FLOAT_CONST'
         { MP_ROM_QSTR(MP_QSTR_e), ULAB_REFERENCE_FLOAT_CONST(ulab_const_float_e) },
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/hgfs6d_16/openmv/openmv­c/omv/modules/ulab/code/numpy/../ndarray.h:71:40: note: (near initialization for 'ulab_numpy_globals_table[4].value')
 #define ULAB_REFERENCE_FLOAT_CONST(id) ((mp_obj_t)(id))

kwagyeman, 2023-04-16:
Hi, you made a forum post about this issue... so closing this.

closed 2023-04-16 by kwagyeman