openmv/openmv #1839: OV7675 (Arduino Nano 33 BLE Sense TinyML kit) detected as OV7690

j054n, 2023-05-10:
**Describe the bug**
The cam connected in Arduino Nano 33 BLE Sense TinyML kit shield gives error one OV7690 instead of OV7675 when connected in OpenMV
**To Reproduce**
Steps to reproduce the behavior:
1. Simply connect in OpenMV
2. In status bar says NANO33 as board and OV7690 as sensor
3. When executes basic example gives first error overflow by QVGA resolution, but when down to QQVGA the viewer shows erroneous image divide in two similar parts with less resolution
4. See error

**Expected behavior**
Correct sensor detected and image in viewer corrected

**Screenshots**
[image]

**Please complete the following information**
 - IDE version: 3.0.3
 - Firmware version: 4.4.2
 - Board: Arduino Nano 33 BLE Sense 
 - OS: Tested on Windows and MacOS

iabdalkader, 2023-05-10:
Only the OV7675 is tested and supported for this board.

j054n, 2023-05-10:
Precisely for this reason it is very strange, this is the card with the camera OV7675, and I have tried more than one similar with the same result.

[image]

I have seen in sources the same ID for OV7670 and OV7690, could be it related?

[image]

Would it be possible to force OpenMV to use that model instead of autodetection?

iabdalkader, 2023-05-10:
I just checked and the OV7690 driver is not enabled for this board, so it is using the OV767x driver, but the IDE might just be showing the wrong sensor ID. 

I don't have this kit with me anymore so can't test it, but if you can build the firmware try setting this to `70`:

```
#define OMV_OV7670_VERSION      (75)
```

j054n, 2023-05-10:
I dont know how can do it.
I would have to compile the whole OpenMV IDE or only Nano33 board firmware? 
In which file do you refer line 70? 
I can see this line in  line 48 of /openmv/openmv/blob/master/src/omv/boards/arduino_nano_33_ble_sense/omv_boardconfig.h

iabdalkader, 2023-05-10:
No just the Nano firmware and yes that's the file to edit.

j054n, 2023-05-11:
I have changed the nano firmware with different values of macros OMV_OV7670_VERSION and OV7690_ID, compiled in docker for target 'arduino_nano_33_ble_sense', and finally loading firmware from OpenMV-Tools-Run Bootloader menu.
The result is the same as before: detects OV7690 with bad images captured.
Any suggest? I can't imagine the problem origin, perhaps it would be necessary identify these board with the second byte for version, because the first is always 0x76.

iabdalkader, 2023-05-12:
You don't need to worry about the OV version shown in the IDE, like I said the only driver enabled for this board is OV7670 (see `OMV_ENABLE_OVxxx` macros), so that's the only driver that will be used in runtime (even if you connect an OV7690).  The IDE just doesn't know how to tell which sensor is actually connected since they all share the same ID, we can possibly fix that but it doesn't really affect anything.

It seems there might be another issue, I don't have a kit to test with will ask for one but it could take some time, if you want to try something, there was a recent change to the registers, you could revert it try:

https://github.com/openmv/openmv/commit/50a3d3efd551a5c06d909938a5ee3a72b7243fae

iabdalkader, 2023-05-25:
Fixed.

closed 2023-05-25 by iabdalkader

j054n, 2023-05-31:
Thank you Ibrahim.
I have finally got the new firmware work in Windows environment, initially I have tried it in Linux without success, do you know if there is any incompatibility so that OpenMV does not detect the device in the usb port?

Aparentely correct output of dmesg command:
[1220953.504488] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
[1220953.669241] usb 1-1.3: USB disconnect, device number 95
[1220955.427552] usb 1-1.3: new full-speed USB device number 96 using ehci-pci
[1220955.538701] usb 1-1.3: New USB device found, idVendor=2341, idProduct=015a, bcdDevice= 1.00
[1220955.538708] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1220955.538711] usb 1-1.3: Product: Board in FS mode
[1220955.538714] usb 1-1.3: Manufacturer: MicroPython
[1220955.538716] usb 1-1.3: SerialNumber: 000000000000
[1220955.539377] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device

Thanks in advance

j054n, 2023-06-01:
The permissions were checked for my user and belong to dialout and plugdev groups.
The /dev/ttyACM0 has +rw rights for all:
crw-rw-rw- 1 root dialout 166, 0 may 31 14:12 /dev/ttyACM0
Its difficult to understand because no problems to access from Arduino IDE, included OV7575 cam

kwagyeman, 2023-06-01:
Mmm, okay, I'm working on updating the IDE to the latest version of Qt 6.5. Once complete this may resolve the issue. I can't tell what is wrong otherwise. From the IDE code I use the same API for windows/linux. So, the issue would either be int he Qt library and/or the operating system settings interfacing with the Qt library.

j054n, 2023-06-02:
I expect you can solve it with next update.
In any case by taking a look to rules file openmv/share/qtcreator/pydfu/50-openmv.rules

```
ATTRS{idVendor}=="1209", ATTRS{idProduct}=="abd1", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="1209", ATTRS{idProduct}=="abd1", ENV{MTP_NO_PROBE}="1"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE:="0666", SYMLINK+="openmvdfu"
KERNEL=="ttyACM*", ATTRS{idVendor}=="1209", ATTRS{idProduct}=="abd1", MODE:="0666", SYMLINK+="openmvcam"
```

the vid/pid included are different from output of lsusb command:

```
$ lsusb
Bus 001 Device 013: ID 2341:015a Arduino SA Board in FS mode
```
Is it correct?

j054n, 2025-07-02:
I'm trying to use the OV7675 camera again and it's being detected as an OV7690. Now a control error appears directly.
Is it no longer compatible?

PR #1850 (2023-05-25): ports/nrf: Fix OV767x on ML kit.
Fixes #1839