openmv/openmv #1849: "socket error:  [Errno 11] EAGAIN" on WiFi shield

cyhcyhgo, 2023-05-25:
Dear developers:
      While I was running MJPEG_Streamer_AP.py on WiFi shield, an error message "socket error:  [Errno 11] EAGAIN" showed in the terminal. The board could not start AP mode. What should I do to fix this problem? Is my WiFi shield broken?

kwagyeman, 2023-10-10:
Hi, is this still an issue?

cyhcyhgo, 2023-10-15:
> Hi, is this still an issue?

Hi. Though the error message is still shown, I can connect the AP and watch the remotely transmitted video from my phone. Once connectting, the error message will disappear.

iabdalkader, 2023-10-27:
The error message is normal, the server socket just times out and restarts. Setting a timeout on the server socket is a workaround for a WINC driver/firmware bug, it has always been this way and it works fine, maybe a host driver update in the future will fix it. As for the AP mode issue, `WEP` security has been deprecated since firmware 19.7.6, that's why it stopped working a while ago. I made some changes and fixes to support using WPA with AP mode, if you try the latest development firmware, AP mode should be working fine now (just remove the `security` arg, and use `ssid`) . Also I'm going refactor all of the networking scripts soon, for all boards, to switch to a common WiFi interface and update all those args.

PR #1997 (2023-10-27): drivers/winc: Fix AP mode.
The `WEP` security mode has been deprecated since firmware 19.7.6 the only two supported modes now are open and WPA.

Fixes #1849
closed 2023-10-27 by iabdalkader