openmv/openmv #1851: HVGA cropping increased between two firmware versions

tejalbarnwal, 2023-05-27:
**Describe the bug**
The crop for HQVGA has increased between v4.4.1 and v4.4.2  and the development version.
The FOV or the area of the image captured should have remained the same, but its not.
The area captured by HQVGA with firmware version v4.4.1 is greater than in v4.4.2 and the development version.

**Expected behavior**
The expected behavior would show as the area captured remains the same irrespective of the firmware version.

**Screenshots**
version 4.4.1
HVGA
[image]

version 4.4.1
HQVGA

version development
HVGA
[image]

version development
HQVGA
[image]

**Please complete the following information**
 - Board: Arduino Nicla Vision
 - OS: Linux and Windows

iabdalkader, 2023-05-27:
I think we had to reduce it to fix the windowing/framerate or something. @kwagyeman

tejalbarnwal, 2023-05-27:
The greater field of view was better, as we could use it instead of HVGA and still have greater framerate(than HVGA)

kwagyeman, 2023-05-31:
Hi, the code does this to keep the frame rate up since we had a user asking if we could make the sensor run faster. When using the HVGA resolution the downsampling amount is 1600/480=3.333333. The code limits this to 3. So, we take a crop of 1440 pixels from 1600 and then scale that down 3x to get 480.

For HQVGA the math would then be 1600/240=6.66666. The code limits this to 3. So, we take a crop of 720 from the center 1600 and then scale that down by 3 which gives 240.

I could allow for a scale of up to 5. So, 1200 pixels cropped by 1600 and then scaled down to 240. However, the frame rate will be halved.

To make this work as the same before I just have to change the 3 here to 5: https://github.com/openmv/openmv/blob/master/src/omv/sensors/gc2145.c#L857

Please let me know your thoughts on this. I could make an IOCTL to allow you to control this number. As of right now, the code is optimized to produce a faster FPS over viewing angle. Unfortunately, it's not possible to have both.

tejalbarnwal, 2023-06-01:
Thank you so much,
Yeah, an IOCTL would be good for future
So as a temporary fix, I would have to go back to the earlier functionality of increased area but reduced FPS, I would just change the ratio to 5, right?

kwagyeman, 2023-06-01:
Yes, if you fork our repo and enable github actions then it will automatically build the firmware for you in the code. Then you can just make a commit to that file by changing the 3 to a 5 and you'll get the behavior you want.

tejalbarnwal, 2023-06-01:
Great, thank you so much!

closed 2023-06-01 by tejalbarnwal

iabdalkader, 2023-06-01:
I was just wondering, @tejalbarnwal wouldn't a wide angle lens be better than reverting this code ?

tejalbarnwal, 2023-06-01:
Yeah, definitely it would be.
Its just we are working with off-the-shelf Nicla Vision and we dont wanna modify it. Further, we have already built on the available functionalities.