openmv/openmv #1854: find_rects the roi parameter to cause the result error

kidswong999, 2023-05-31:
```
import sensor, image, time

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.VGA)
sensor.skip_frames(time = 2000)
clock = time.clock()

while(True):
    clock.tick()
    img = sensor.snapshot()
    img.draw_rectangle(0,0,640,480,(255,255,255),0,fill=True)
    
    img.draw_rectangle(100,100,50,50,(0,0,0),0,fill=True)
    img.draw_rectangle(540,100,50,50,(0,0,0),0,fill=True)
    img.draw_rectangle(100,380,50,50,(0,0,0),0,fill=True)
    img.draw_rectangle(540,380,50,50,(0,0,0),0,fill=True)
    
    #img.draw_rectangle(500,50,140,200,(0,0,0),2,fill=False)
    
    rects =  img.find_rects(threshold = 1000)
    print('all', rects)

    rects =  img.find_rects(roi=(500,50,130,200), threshold = 1000)
    print('right top corner', rects)

```

I run this code and will only print 

```
all [{"x":540, "y":380, "w":50, "h":50, "magnitude":205726}, {"x":100, "y":100, "w":50, "h":50, "magnitude":205726}, {"x":540, "y":100, "w":50, "h":50, "magnitude":205726}, {"x":100, "y":380, "w":50, "h":50, "magnitude":205726}]
right top corner []
```

expected:

right top corner will have one object.

kwagyeman, 2023-05-31:
Will fix after IDE update.

kwagyeman, 2024-01-03:
Thanks for finding this issue. Sorry for the delay in fixing it.

closed 2024-01-06 by iabdalkader