openmv/openmv #1870: can.recv(0, lst) throws ValueError exception with Cam H7, but not with Cam H7 Plus

PatrickM3, 2023-06-22:
The following code works on the Cam H7 but not on the Cam H7 Plus:

buf = bytearray(8)
lst = [0, 0, 0, memoryview(buf)]
ret = can.recv(0, lst)

IDE 3.0.3

kwagyeman, 2024-01-03:
Will address when I get home.

kwagyeman, 2026-09-30:
This is by design, not a bug. `pyb.CAN.recv(fifo, list)` requires a 5 element list: `[id, is_extended_id, is_rtr, fmi, data]`, with the `memoryview` at index 4. A 4 element list raises `ValueError`. The H7 that accepted it was running older firmware.

Verified on current master on an H7 in loopback: 4 element list raises `ValueError`, 5 element list and `recv(0)` with no list both return the frame.

The recommended API is now `machine.CAN`. There the call is `can.recv(lst)` with `lst = [0, memoryview(buf), 0, 0]` and the result is `[id, data, flags, errs]`. Closing.

closed 2026-09-30 by kwagyeman