openmv/openmv #1871: Arduino Nicla Vision unable to make MQTT connections

shan716, 2023-06-23:
**Describe the bug**
When running the MQTT_pub.py example under Arduino Boards -> Nicla Vision -> Wi-Fi, the program only allows a secure connection to 'test.mosquitto.org'. When connecting to any other brokers, the following exception gets thrown:
```
Traceback (most recent call last):
  File "<stdin>", line 28, in <module>
  File "mqtt.py", line 134, in connect
MQTTException: 2
sdio_transfer_cmd53: timeout wr=1 len=128 dma=1 buf_idx=0 STA=00000040 SDMMC=00000000:00000c22 IDMA=00000001
[CYW43] send_ethernet failed: -110
bad RESP1: 53 2800
sdio_transfer_cmd53: error=ffffffff wr=0 len=4 dma=0 buf_idx=0 STA=00400000 SDMMC=00000000:00000c60 IDMA=00000001
```

**To Reproduce**
1. Run MQTT_pub.py on Arduino Nicla Vision
2. Use a broker other than 'test.mosquitto.org'

**Expected behavior**
Connection to specified MQTT broker. 

Detailed Explanation: https://forums.openmv.io/t/issues-running-mqtt-with-nicla-vision/8478 
The program should be updated to follow the format of `rtsp.py`

 - IDE version: 4.0
 - Firmware version: 4.4.2
 - Board: Arduino Nicla Vision
 - OS: iOs and Linux

kwagyeman, 2023-06-23:
TODO: I need to robustify the script.

iabdalkader, 2023-06-28:
This might be a different issue, you can see there are SDIO errors that shouldn't happen, so I'll look into it.

shan716, 2023-06-28:
@iabdalkader Thank you for your support. I have uploaded my OpenMV to version 4.4.3 and loaded the latest firmware_ARDUINO_NICLA_VISION/firmware.bin onto my board. I no longer see the SDIO errors, but still cannot connect to brokers other than `test.mosquitto.org`. This is the error I'm getting when I run mqtt_pub.py:
```
Traceback (most recent call last):
  File "<stdin>", line 28, in <module>
  File "mqtt.py", line 134, in connect
MQTTException: 2
```

iabdalkader, 2023-06-28:
That's a different issue, the connection in the example is Not secure, you should pass `ssl=True` and remove the port it will use the SSL port for MQTT, see https://github.com/openmv/openmv/blob/master/scripts/libraries/mqtt.py#L45

I don't know how good this mqtt client is with secure connections, if it doesn't work you can try this one:

https://github.com/arduino/arduino-iot-cloud-py/blob/main/src/arduino_iot_cloud/umqtt.py

closed 2023-06-28 by iabdalkader

shan716, 2023-06-28:
Okay, thank you for your help!

iabdalkader, 2023-06-30:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/issues-running-mqtt-with-nicla-vision/8478/6

shan716, 2023-06-30:
Hi iabdalkader, thank you for the follow-up. I posted the issue in the OpenMV forum, and it looks like kwagyeman has plans to eventually robustify this program.

OlisCode, 2023-07-08:
I've also had a lot of the same "MQTTException: 2" problems using the library with mosquitto as a broker.
The Exception code 2 I've found very misleading from mosquitto as it states "client identifier not allowed by server".
I was able to get MQTTworking by:
1. Using  the current development release (not entirely sure if it is needed)
2.  waiting for the Nicla to actually get an IP assigned before doing anything with MQTT (not entirely sure if it is needed):

```
wlan = network.WLAN(network.STA_IF)
wlan.deinit()
wlan.active(False)
wlan.active(True)
wlan.connect(SSID, KEY)
timeout_time = time.time() + 10
print("Connected, waiting for IP")
# Wait until we get a valid IP via DHCP
while(wlan.status() != 3): # STAT_GOT_IP seems to mean 3. I was unable to find STAT_GOT_IP to properly compare against it.
    if time.time() > timeout_time:
        raise Exception("No ip received in timeout time")
# We should have a valid IP now via DHCP
print("WiFi Connected ", wlan.ifconfig())
# Start with the MQTT initialization
```

4. The remaining thing that solved my issues was to set the keepalive to 60, I assume any value other than 0 should work. If the keepalive is set to 0 a mosquitto server with a fresh configuration refuses the connection and therefore causes the "MQTTException: 2".
```
client = MQTTClient("clientID", "hostname", port=1883, ssl=False, keepalive=60)
or
client = MQTTClient("clientID", "hostname", port=1883, user="username", password="password", ssl=False, keepalive=60)
```
As you can see I did not use ssl and therefore might not have come across any other roadblocks that might occur  with using ssl.
The MQTTException: 2 therefore to me seems not to be an issue with OpenMV but with mosquitto.
It would however be great if OpenMV could include more information in the MQTTExceptions and/or set the default value of keepalive to 60 or similar.
In #1892 there is also a mention of umqtt.py from Arduino, this has the same connection problem to mosquitto.
If there is a need to check where a Exception comes from [here](https://github.com/openmv/openmv/blob/master/scripts/libraries/mqtt.py) you can find the source for mqtt.py to check yourself.

On a sidenote if you get this error:
```
Traceback (most recent call last):
  File "<stdin>", line 39, in <module>
  File "mqtt.py", line 91, in connect
OSError: -2
```
The Hostname could not resolve.

I hope this will be useful to someone as it took me nearly a day to figure out until at some point I ended up in wireshark.😄

iabdalkader, 2023-07-08:
Yes it is very helpful, thank you.

> * Using  the current development release (not entirely sure if it is needed)
> * waiting for the Nicla to actually get an IP assigned before doing anything with MQTT (not entirely sure if it is needed):

Both are needed, although you should just call `wlan.isconnected()` instead of `status()` because it's supported by all other networking interfaces, `status()` could have different meaning. Note I already updated all scripts to poll the connection.

> 4\. If the keepalive is set to 0 a mosquitto server with a fresh configuration refuses the connection

@shan716 Could this be the issue with your MQTT server ?