openmv/openmv #1875: Target PICO and ARDUINO_NANO_RP2040_CONNECT compile fail.

overheat, 2023-06-25:
**Describe the bug**
Eather `PICO` or `ARDUINO_NANO_RP2040_CONNECT` compile fail on `master` branch. 
 
**To Reproduce**
Steps to reproduce the behavior:
`make -j$(nproc) TARGET=ARDUINO_NANO_RP2040_CONNECT -C src`
or
`make -j$(nproc) TARGET=PICO -C src`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
```
[  5%] Linking CXX executable firmware.elf
/usr/local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/firmware.dir/home/ubuntu/openmv/src/micropython/py/objmodule.c.obj:(.rodata.mp_builtin_module_table+0x54):
undefined reference to `mjpeg_module'
```
**Please complete the following information**

 - Board: PICO/ARDUINO_NANO_RP2040_CONNECT
 - OS: Ubuntu 22.04
 - Arm64

iabdalkader, 2023-06-28:
It must be a local issue, you can see the CI is building all the targets fine:

https://github.com/openmv/openmv/actions/runs/5392123197

closed 2023-06-28 by iabdalkader

overheat, 2023-06-28:
@iabdalkader 
yes, it compiled on x86 cpu aws cloud server.
Where can I find the guide on arm64 server?

iabdalkader, 2023-06-28:
> Where can I find the guide on arm64 server?

There isn't any, I never built the firmware on ARM before, if you do, you're welcome to send a PR to add one.