openmv/openmv #1911: The tf.regression() function outputted exactly same results while the input changed different values.

VisionShilin, 2023-08-14:
The tf.regression() function outputted exactly same results while the input changed different values.

kwagyeman, 2023-08-15:
Yeah, it depends on what model you are running.

VisionShilin, 2023-08-16:
The model I used outputted correct results in the TF training environment and did outputted different results using various inputs. It meant the model worked correctly. But when I load the model into OpenMV using tf.regression(), it always output exactly same results whenever I changed any inputs. Is tf.regression() really works correctly? Or need any license?

kwagyeman, 2023-08-18:
No, it's free to use. It's user contributed code though. I don't believe it has bugs but you can check what is going on here:

https://github.com/openmv/openmv/blob/master/src/omv/modules/py_tf.c#L263

https://github.com/openmv/tensorflow-lib/blob/master/libtf.cc#L377

The author of the code reported it worked still.

kwagyeman, 2024-01-03:
@VisionShilin Hi, I have time to deal with this bug along with general libtf updates. Can you send me your regression model to test with? I don't have anything do use to compare with right now as this was a user submitted feature.

VisionShilin, 2024-01-04:
@kwagyeman I gave up using OpenMV and used the TF c lib directly.

kwagyeman, 2024-01-04:
@VisionShilin do you have the tflite file and input data? I can use that to fix the issue if I know what the expected output should be.

This wouldn't be specific to OpenMV.

kwagyeman, 2024-01-23:
Closing this as there's no new info.

kwagyeman, 2024-01-23:
Note: https://github.com/openmv/openmv/pull/2108.

We've taken ownership of the regression method. There were some issues with the code before.

closed 2024-01-23 by kwagyeman

saricaBugrahan, 2024-01-24:
Does it still the case? I have similar results regardless of the input array. The firmware version is 4.5.2(which IDE says it's the latest.). Additionally, I have tested with 3 different models. Nevertheless, I cannot fix the getting same output while input changing.

kwagyeman, 2024-01-24:
PR 2108 should fix the issue. From what I can tell, if you didn't pass an NDARRY from ulab the code was just just treaing any micropython object as that and grabbing garbage.