openmv/openmv #1926: RPC Library SPI bus transfer is not working

kwagyeman, 2023-09-09:
Schedule to fix this when doing the update to convert the RPC library to the machine module.

iabdalkader, 2023-11-22:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/spi-slave-error-no-attribute-put-short-timeout/8966/4

kwagyeman, 2023-11-27:
This post goes into detail why this is bugged right now. https://forums.openmv.io/t/spi-deinit-changing-pin-function/8972

Basically, the I/O pin gets deinited by the spi bus and needs to be reinited. This should fix the issue. Not sure if I should make the SPI code though not deinit the SS pin. This is the behavior we want in C code... just not in python land.

@iabdalkader Any thoughts on this? I can just fix the issue in the RPC library easily. Or, should we not deinit the SS pin in the HAL deinit call?

closed 2024-01-03 by kwagyeman