openmv/openmv #1927: at removable disk(E:) the main.py, readme and openmv_disk file is presented . I tried openmv ide software and clicked to run it showing Run time error: sensor control failed. how to solve this problem?

bsri672, 2023-09-13:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Please complete the following information**
 - IDE version: [e.g. 2.6.8]
 - Firmware version: [e.g. 3.8.0]
 - Board: [e.g. OPENMV-H7+/Portenta]
 - OS: [e.g. Windows/Linux/iOs]

bsri672, 2023-09-13:
at removable disk(E:) the main.py, readme and openmv_disk file is presented . I tried openmv ide software and clicked to run it showing Run time error: sensor control failed. how to solve this problem? using open mv cam h7 plus

kwagyeman, 2023-09-13:
Hi, is the sensor secure on the camera unit? That error happens when we are not able to initialize the sensor module.

bsri672, 2023-09-14:
How to initialize the sensor module. lepton 3.5 openmv h7 plus camera?

kwagyeman, 2023-09-14:
It should just work if you are using the flir lepton adapter.

bsri672, 2023-09-14:
how to solve this error ? sensor control failed: runtime error

bsri672, 2023-09-14:
how to mount internal drive files. in E drive only main.py, readme, and openmv_disk is there but if i click on run it is not running.

iabdalkader, 2023-09-14:
Hi, please use the forums to ask any questions you have https://forums.openmv.io/

closed 2023-09-14 by iabdalkader