openmv/openmv #1929: how to mount internal drive files. in E drive only main.py, readme, and openmv_disk is there but if i click on run it is not running.

bsri672, 2023-09-14:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Please complete the following information**
 - IDE version: [e.g. 2.6.8]
 - Firmware version: [e.g. 3.8.0]
 - Board: [e.g. OPENMV-H7+/Portenta]
 - OS: [e.g. Windows/Linux/iOs]

bsri672, 2023-09-14:
how to mount internal drive files. in E drive only main.py, readme, and openmv_disk is there but if i click on run it is not running. ide version:4.0.1, firmware version: 4.4.3, board: openmv cam h7 plus stm32h743 w/sdram os:windows

iabdalkader, 2023-09-14:
Hi, please use the forums to ask any questions you have https://forums.openmv.io/

closed 2023-09-14 by iabdalkader