openmv/openmv #1989: Arduino Nano 33 BLE Sense rev1 cannot import lsm9ds1 or apds9960

lkjell, 2023-10-26:
**Describe the bug**
Examples for Arduino Nano 33 BLE Sense rev 1 which requires module lsm9ds1 or apds9960 does not work.

**To Reproduce**
Steps to reproduce the behavior:
Flash the latest Firmware (4.5.1 or 4.5.0 stable).
Run the example from: https://github.com/openmv/openmv/tree/master/scripts/examples/10-Arduino-Boards/Nano-33-BLE-Sense/01-Sensors

**Expected behavior**
For IMU gyroscope, accelerometer and magnetometer does not work for example.
Firmware 4.4.3 work with IMU but with the new import i.e. ` import imu`.

**Screenshots**

**Please complete the following information**
 - IDE version: 4.0.14
 - Firmware version: 4.5.1
 - Board: Arduino Nano 33 BLE Sense rev1
 - OS: Windows 11

iabdalkader, 2023-10-26:
Which example did you run ? I tested those recently and they were working. You should be using `imu` module which abstracts the acc/gryo chipset to work on REV1/2.

lkjell, 2023-10-26:
```
# IMU example for Arduino Nano BLE 33 Sense (REV1 and REV2).

import time
import imu
from machine import Pin, I2C

bus = I2C(1, scl=Pin(15), sda=Pin(14))
imu = imu.IMU(bus) # Problem with this line

while (True):
    print('Accelerometer: x:{:>8.3f} y:{:>8.3f} z:{:>8.3f}'.format(*imu.accel()))
    print('Gyroscope:     x:{:>8.3f} y:{:>8.3f} z:{:>8.3f}'.format(*imu.gyro()))
    print('Magnetometer:  x:{:>8.3f} y:{:>8.3f} z:{:>8.3f}'.format(*imu.magnet()))
    print("")
    time.sleep_ms(100)

```

```
from time import sleep_ms
from machine import Pin, I2C

from apds9960 import uAPDS9960 as APDS9960

bus = I2C(1, sda=Pin(13), scl=Pin(14))
apds = APDS9960(bus)

apds.setProximityIntLowThreshold(50)

print("Proximity Sensor Test")
print("=====================")
apds.enableProximitySensor()

while True:
    sleep_ms(250)
    val = apds.readProximity()
    print("proximity={}".format(val))
```

iabdalkader, 2023-10-26:
Ah yes sorry about that, some modules were dropped.

closed 2023-10-26 by iabdalkader

lkjell, 2023-10-26:
Thanks :-) Fast.

iabdalkader, 2023-10-26:
If you test the dev firmware now they should both be working, also the examples for APDS9960 were using wrong pins the bus should be:
```Python
bus = I2C(1, scl=Pin(15), sda=Pin(14))
```

PR #1990 (2023-10-26): boards/arduino_nano_33_ble_sense: Fix frozen drivers.
Fixes #1989