openmv/openmv #2038: Screen stuttering on GIGA R1 since 4.5.1

Bexin3, 2023-11-30:
**Describe the bug**
Stuttering on GigaR1 using display.py example

**To Reproduce**
Hello, on both 4.5.1 and 4.5.2 there is stuttering on the screen using either OV5640 or OV7690.
I wasn't able to reproduce the issue on 4.5.0 using OV7690, while I don't have drivers for this version where OV5640 would be enabled so I can't test that.
sensor.set_framebuffers(1) fixes it, however decreases performance - and was not needed with older versions after all.

**Please complete the following information**
 - IDE version: 4.0.14
 - Firmware version: 4.5.1+
 - Board: GIGA R1
 - OS: MacOS

iabdalkader, 2023-11-30:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/adafruit-ov-5640-not-recognised-on-giga-r1/8945/27

Bexin3, 2023-11-30:
> @Bexin3 - can you provide the demo script.

Just to clarify, do you mean the code? If so, its;
```
# LCD, touch panel and camera example.

import sensor
import time
import display
from gt911 import GT911
from machine import I2C

IMG_OFFSET = 80
touch_detected = False
points_colors = ((255, 0, 0), (0, 255, 0), (0, 0, 255), (0, 255, 255), (255, 255, 0))

sensor.reset()  # Reset and initialize the sensor.
sensor.set_pixformat(sensor.RGB565)  # Set pixel format to RGB565 (or GRAYSCALE)
sensor.set_framesize(sensor.VGA)  # Set frame size to QVGA (320x240)
sensor.set_vflip(True)  # Flip image for the display
# sensor.set_transpose(True) # Hardware transpose will be slower at this resolution

lcd = display.DSIDisplay(
    framesize=display.FWVGA, portrait=True, refresh=60, controller=display.ST7701()
)

# Note use pin numbers or names not Pin objects because the
# driver needs to change pin directions to reset the controller.
touch = GT911(
    I2C(4, freq=400_000),
    reset_pin="PI2",
    irq_pin="PI1",
    touch_points=5,
    refresh_rate=240,
    touch_callback=lambda pin: globals().update(touch_detected=True),
)

# Create a clock object to track the FPS.
clock = time.clock()

while True:
    clock.tick()  # Update the FPS clock.

    # Capture a new frame
    img = sensor.snapshot()

    # Draw touch points if touch was detected.
    if touch_detected:
        n, points = touch.read_points()
        for i in range(0, n):
            img.draw_circle(
                points[i][0] - IMG_OFFSET,
                points[i][1],
                points[i][2] * 3,
                points_colors[points[i][3]],
                thickness=2,
            )
        touch_detected = False

    # Rotate the image in place.
    img.replace(transpose=True)

    # Draw the image on the display.
    lcd.write(img, y=IMG_OFFSET)

    print(clock.fps())
```

kwagyeman, 2023-11-30:
Hi, with the latest firmware

`img.replace()` is not required anymore. You can do:

`lcd.write(img, hint=img.CENTER | img.ROTATE_90 | img.SCALE_ASPECT_KEEP)`

This will cut out two copies from SDRAM which should improve performance.

kwagyeman, 2023-11-30:
I'm running this with my Pure Thermal without any display issues:

`import sensor
import display
import image

sensor.reset()  # Initialize the camera sensor.
sensor.set_pixformat(sensor.RGB565)  # or sensor.GRAYSCALE
sensor.set_framesize(sensor.VGA)  # Special 128x160 framesize for LCD Shield.
lcd = display.RGBDisplay()
lcd.backlight(1)

while True:
    lcd.write(sensor.snapshot() , hint=image.CENTER | image.SCALE_ASPECT_KEEP)`

Bexin3, 2023-11-30:
> Hi, with the latest firmware
> 
> `img.replace()` is not required anymore. You can do:
> 
> `lcd.write(img, hint=img.CENTER | img.ROTATE_90 | img.SCALE_ASPECT_KEEP)`
> 
> This will cut out two copies from SDRAM which should improve performance.

I will try this. Any idea regarding the stuttering though, as it's only a thing with new firmwares (I have also tested official ones with OV5640 disabled to make sure its not my fault with the code.

kwagyeman, 2023-11-30:
I get some suttering with this on the Pure Thermal:

```
import sensor
import display
import image

sensor.reset()  # Initialize the camera sensor.
sensor.set_pixformat(sensor.RGB565)  # or sensor.GRAYSCALE
sensor.set_framesize(sensor.VGA)  # Special 128x160 framesize for LCD Shield.
lcd = display.RGBDisplay()
lcd.backlight(1)

while True:
    lcd.write(sensor.snapshot() , hint=image.CENTER | image.ROTATE_90 | image.SCALE_ASPECT_KEEP)  # Take a picture and display the image.
```

kwagyeman, 2023-11-30:
Uh, I don't get any stuttering with the new code unless I add the rotate... This increases processor load. However, I designed it to do as much work in SRAM as possible to avoid huge memory bandwidth issues.

Bexin3, 2023-11-30:
> Uh, I don't get any stuttering with the new code unless I add the rotate... This increases processor load. However, I designed it to do as much work in SRAM as possible to avoid huge memory bandwidth issues.

Does the stuttering happen on lower resolutions for you? It does for me. Also could it be something with rotate method changing, as it didn't happen before

kwagyeman, 2023-11-30:
As a test remove the rotate. This is a pure processor operation that adds load on the system bus. If the screen clears up then it's a bandwidth issue as I mentoned.

kwagyeman, 2023-12-01:
> Does the stuttering happen on lower resolutions for you? It does for me. Also could it be something with rotate method changing, as it didn't happen before.

I don't get stuttering at QQVGA. At QVGA and VGA I do.

Bexin3, 2023-12-01:
> > Does the stuttering happen on lower resolutions for you? It does for me. Also could it be something with rotate method changing, as it didn't happen before.
> 
> 
> 
> I don't get stuttering at QQVGA. At QVGA and VGA I do. It's only every 30 seconds or so for me on the Pure Thermal.

Just to check, do you also get this on 4.5.0 fw?

kwagyeman, 2023-12-01:
It's the same for me on v4.5.0.

As just a test. Can you turn off the frame buffer on the IDE and see if the glitches go away? I don't get them anymore once you remove the JPEG compression load.

iabdalkader, 2023-12-06:
I can't reproduce this issue, at least not with the OV7675. The display seems to be working perfectly fine.

kwagyeman, 2024-01-15:
I can't reproduce this on the latest. It seems fine.

closed 2024-01-15 by kwagyeman