openmv/openmv #2049: HID mode doesn't work after new firmware update

Igino95, 2023-12-14:
Hello, 

i'm working with Arduino nicla vision, openMV IDE 4.1.1 on windows. Frome serial terminal after connection: 
OpenMV 5f9fe8b; MicroPython b0b6a67; NICLAVISION-STM32H747

After connecting the board to the PC via USB cable, the open MV IDE told me to update the firmware, I did it and now the HID mode no longer works.

I also tried resetting the board and loading the example file, also creating the boot.py file like this:

import pyb
pyb.usb_mode('VCP+HID') # serial device + mouse (UNCOMMENT THIS LINE!)

and saving this script as main.py:

import pyb
import time

hid = pyb.USB_HID()

while True:
     # x, y and scroll
     # move 10 pixels to the right
     hid.send((0, 10, 0, 0))
     time.sleep_ms(500)

but the board keeps crashing and restarting

Can you help me? do you have the old firmware?

Thank you

kwagyeman, 2023-12-14:
The boot process was changed around in the latest firmware update. @iabdalkader will know more about this.

The old firmware is here: https://github.com/openmv/openmv/releases/tag/v4.5.0

Igino95, 2023-12-15:
Ah ok, will it be available again?

In any case the old version works, thank you.

iabdalkader, 2023-12-19:
>> Ah ok, will it be available again?

I will try to make it work again.

Igino95, 2023-12-21:
Ok thank you

iabdalkader, 2023-12-23:
The only way to fix this is to move boot.py before the USB device is initialized, does this break anything for you @kwagyeman ?

Alternatively, this in `boot.py` (or `main.py`) works too:

```Python
import pyb

if pyb.usb_mode() == "VCP+MSC":
    pyb.usb_mode(None)
    pyb.usb_mode("VCP+HID")
```

kwagyeman, 2023-12-23:
It's not change the boot order.

I think the fix you mentioned in code is enough.

Thanks! :)

Igino95, 2023-12-27:
Ok, thank you for support! :)

closed 2023-12-27 by Igino95