openmv/openmv #2070: Debug Issue with jpeg file data causing some desktop programs not to be able to handle it.

kwagyeman, 2024-01-02:
Our JPEG images have some issue with their data that causes issues. Determine what that is and fix it.

kwagyeman, 2024-01-02:
On comparison with our STM32 JPEG compressor and the OV5640 (which displays correctly on desktop programs), there is no difference in the JPEG headers. So, the issue is somewhere in the scanline data. Our software compressor has the same issue as the STM32 one... FFMEG throws an error about extra bits. So, will hunt down this now.

[image]
[image]

kwagyeman, 2024-01-02:
MJPEG video files created by the OV5640 or our jpeg compressor are treated the same by FFMPEG and VLC. They do not have errors. So, the issue must be related to our programs handle JPEG RTSP streams.

So, I would say our jpeg file format is fine. This is an RTSP specific issue.

kwagyeman, 2024-01-02:
Okay, determined the issue.

Only FFMPEG can play GRYSCALE MJPEG streams and YUV444/YUV422/YUV420 RGB streams. Other players like VLC can only handle YUV422 RGB Streams (not YUV420).

The reason the OV5640 works versus our JPEG is that does YUV422 JPEGs versus YUV444 JPEGs.

...

Note, VLC can decode YUV444/Grayscale/YUV420 JPEGs fine when reading an MJPEG file video. However, it cannot handle them in an RTSP stream.

...

Not sure what other players problems are, but, it's probably the same issue. They lack code to handle anything but YUV422 RGB which is the most common format.

...

What is the fix for this? I guess we can default JPEG compression to YUV422. But, also add some setting maybe to force YUV422 when necessary.

iabdalkader, 2024-01-02:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/rt1062-bugs-in-examples-april-tag-mjpeg-streamer-rtsp-video-streamer-wlan/9036/8

kwagyeman, 2024-01-03:
Will wait on hardware jpeg to be merged before sending a PR to handle this.