openmv/openmv #2081: Autofocus not working reliably on iMX-RT1060 using FPC camera

kenobi986, 2024-01-04:
Auto focus using FPC OV5670 camera on the iMX-RT1060 was not working. I enabled the auto focus feature in omv_boardconfig.h and compiled the bin file and flashed into the board. I was not able to get focus at various distances for some reason. Also, I am wondering if you've examples for the auto-focus (this might be an enhancement). 

I raised the issue on the forums and here it is: 

https://forums.openmv.io/t/autofocus-not-working-reliably-on-imx-rt1060-using-fpc-camera/9061

[image]

kwagyeman, 2024-01-04:
Thanks, I'll debug this when I get home next week.

kwagyeman, 2024-01-13:
Hi, I just tested this feature out. It seems to work. Note that the MCU on the camera decides what focus level to use. Not us. I can't comment on it's results of focusing.

```
import sensor, image, time

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time = 2000)

sensor.ioctl(sensor.IOCTL_TRIGGER_AUTO_FOCUS)
clock = time.clock()

while(True):
    clock.tick()
    img = sensor.snapshot()
    
    print(clock.fps())
```

This script above results in the system focusing the lens after 2 seconds. I held an item with a barcode in front of it and it did focus as expected.

You can execute sensor.ioctl(sensor.IOCTL_RESET_AUTO_FOCUS) to release the settings or sensor.ioctl(sensor.IOCTL_TRIGGER_AUTO_FOCUS) to focus again.

You cannot call the function in a loop though. The camera can't just continually auto focus. It's a one shot thing you can call every now and then.

kwagyeman, 2024-01-13:
As for wait on auto focus... that doesn't seem to be that useful.

read reg 0x3029 and see if it's:

0x70 -> IDLE
0x00 -> Focusing
0x10 -> Focused

You'll see it go from 0 to 0x10. The default value will be 0x70.= before anything is executed.

closed 2024-01-20 by iabdalkader