openmv/openmv #2118: Fix ADC Ports on Nicla

kwagyeman, 2024-01-26:
This user tried to use the ADC ports on the Nicla: https://forums.openmv.io/t/how-to-use-a1-and-a2-ports-with-adc/9164

However, it appears PF13 doesn't work. I got PF3 to work using the machine module versus pyb.

Anyway, I googled around and found this: https://github.com/micropython/micropython/pull/7407

So, I guess you already tried to fix this and it wasn't accepted? Not sure what's going on. I also found this: https://github.com/micropython/micropython/issues/4435

Can this be fixed for the Nicla in our firmware?

iabdalkader, 2024-01-26:
> Anyway, I googled around and found this: [micropython/micropython#7407](https://github.com/micropython/micropython/pull/7407)
> 
> So, I guess you already tried to fix this and it wasn't accepted?

That was merge 3 years ago.

Nicla's analog pins shouldn't have any problems. The only remaining issue related to ADC is dual pads pins, which we're going to fix soon.

kwagyeman, 2024-01-26:
This is fixed in the machine module. Do not use pyb.

closed 2024-01-26 by kwagyeman