openmv/openmv #2183: Fix OV5640 Lens Shading

kwagyeman, 2024-03-17:
Fix the green dot in the OV5640 image.

iabdalkader, 2024-03-17:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/spot-in-image-not-homogeneous/9344/2

kwagyeman, 2024-04-03:
Found a potential fixed and posted it to forums. Waiting on customer feedback.

kwagyeman, 2025-02-08:
This is not fixable. There's a CRA mismatch between the lens and the OV5640. The green center is an artifact of that. Messing around with lens shading can help but not fix.

[image]

[image]

closed 2025-02-08 by kwagyeman