openmv/openmv #2193: Using GC2145 on Arduino GIGA

cyhcyhgo, 2024-04-13:
Hello! I flashed v4.5.3 firmware on my Arduino GIGA board since this version supports GC2145. But when I plug in GC2145 module(bought from ArduCam), the IDE cannot detect the camera. Does there still have problem on drivering GC2145? 
[image]
[image]

cyhcyhgo, 2024-04-13:
"无"means "None of sensors have been detected."

kwagyeman, 2024-04-13:
Hi, the sensor support is not enabled by default. You'd need to enable it in our firmware. Arduino isn't selling that camera module on their store so it's not on in our firmware.

cyhcyhgo, 2024-04-13:
> Hi, the sensor support is not enabled by default. You'd need to enable it in our firmware. Arduino isn't selling that camera module on their store so it's not on in our firmware.

Thanks for your reply. Referring from "omv_boardconfig.h", it seems like the GC2145 is supported in Arduino GIGA. Shall I add extra code to enable it?
[image]

kwagyeman, 2024-04-13:
Yeah, you just need to turn that 0 to 1 and then recompile and load the new firmware.

kwagyeman, 2024-04-13:
Oh! It's already enabled.

Mmm.

cyhcyhgo, 2024-04-13:
> Oh! It's already enabled.
> 
> Mmm.
> 
> The OV5640 being enabled may conflict with it. Try turning that camera off in the config file.

Thanks very much. I'll try it later. But why does OV5640 conflict with GC2145?

kwagyeman, 2024-04-13:
I think they have the same I2C address. Ibrahim added some code though to handle this and figure out which one is attached when this is the case.

I'll have to check what's happening exactly later and I can provide an update.

kwagyeman, 2024-04-14:
Yeah, Ibrahim already added code to fix this issue. It should work.

@iabdalkader - Any thoughts?

cyhcyhgo, 2024-04-15:
> Yeah, Ibrahim already added code to fix this issue. It should work.

Thanks for your work. But I flashed the latest development firmware(4.5.4), the problem still exist.

iabdalkader, 2024-04-15:
Yes this issue has been fixed and someone else tested it and confirmed it was fixed. I have a Giga for testing and some OV7x modules, but I don't have that external GC2145 module to test it myself.

I did try the same sensor config with the Nicla Vision (which has an on-board GC2145) and it detects it with the OV5640 enabled:

```
// Image sensor drivers configuration.
#define OMV_OV5640_ENABLE                   (1)
#define OMV_OV5640_AF_ENABLE                (1)
#define OMV_OV5640_PLL_CTRL2                (0x90)
#define OMV_OV5640_PLL_CTRL3                (0x13)
#define OMV_OV5640_XCLK_FREQ                (12000000)

#define OMV_OV7670_ENABLE                   (1)
#define OMV_OV7670_VERSION                  (75)
#define OMV_OV7670_CLKRC                    (0)

#define OMV_GC2145_ENABLE                   (1)
```

Whatever is wrong with your board must be a different issue.

cyhcyhgo, 2024-04-15:
> Whatever is wrong with your board must be a different issue.

Yes, the Nicla Vision could detect another sensor, but GIGA could not detect any sensor.

kwagyeman, 2024-04-17:
Closing this since it's not an issue with our software it seems.

closed 2024-04-17 by kwagyeman