openmv/openmv #2196: OpenMV RT cannot connect to USB after first booting via 3V3 Vin.

antonvh, 2024-04-14:
**Describe the bug**
OpenMV IDE shows 'No OpenMV Cams Found!' if I connect my OpenMV RT to USB after first booting via 3V3 Vin.

Full project here. Scroll down to troubleshooting section.
https://www.antonsmindstorms.com/2024/04/06/robot-ball-kicker-project-building-a-lego-spike-prime-robot-with-openmv-vision/

**To Reproduce**
Steps to reproduce the behavior:
1. Boot Cam via 3v3 Vin
2. Click on 'Connect (cmd-E)'
3. See error

**Expected behavior**
Always connect. I used a H7 in the exact same setup, with the same scripts, and had no problems connecting.
First connecting to USB and the Vin also solves the issue.

**Please complete the following information**
 - IDE version: 4.1.3
 - Firmware version: OpenMV v4.5.3; MicroPython v1.22-omv.r17; OpenMV IMXRT1060-MIMXRT1062DVJ6A
 - Board: OpenMV RT
 - OS: Mac

[image]

iabdalkader, 2024-04-14:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/usb-c-no-openmv-cam-found-on-openmv-rt-after-booting-with-3v3/9428/3

kwagyeman, 2024-04-14:
Hi, I explain the solution on the forums. Forcing 3.3V into the system bypasses it's power concept. However, it will still work.

You'll need to fiddle with the power button though. The processor does not like the 3.3V rail turning on before it's battery backup rail.

closed 2024-04-14 by kwagyeman