openmv/openmv #2200: Frame Capture Timing Out

kwagyeman, 2024-04-17:
https://forums.openmv.io/t/frame-capture-randomly-timing-out/9482/9

Need to investigate this issue.

justfone, 2024-05-31:
For info, I'm experiencing the same issue as reported in the forum post, almost identical situation as below : 

- RT1062 running on battery or USB
- Wake from deep sleep (time or external trigger)
- Look for QR codes for 5 seconds
- Dump details of any QR found out via UART
- Back to deep sleep

Works perfectly for random periods of time and then starts throwing "Frame Capture TimeOut" exception, if the exception is handled and system is put back into deep sleep then starts working fine again on next wake up.

Initial sensor config is as below

```
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.VGA)
sensor.set_auto_gain(False)
sensor.set_framebuffers(3)
```

kwagyeman, 2024-06-01:
[firmware.zip](https://github.com/user-attachments/files/15520099/firmware.zip)

Can you test out this firmware?

iabdalkader, 2024-06-01:
It's been a while since you reported this issue. I just released a new firmware, v4.5.5, last week, with some important fixes to FB cache. Can you please test the stable v4.5.5 see if it fixes the issue before the other custom firmware above ? Also if you can, please post a script that we can use to reproduce the issue.

justfone, 2024-06-03:
I didn't report the initial issue, I just came across it when I experienced the same problem, I can confirm though that we updated to 4.5.5 last week and were still experiencing it. 

I'm busy on another project for a few days, but I will give the custom firmware a try and generate some minimal code that produces the issue once I get back on this later in the week

kwagyeman, 2024-06-03:
Thanks, this is P0 debugging for us.

justfone, 2024-06-05:
I've tried the custom firmware you provided (4.5.6) but unfortunately the exception is still thrown after a period of time (random, can be minutes or hours)

This is the minimal code we stripped back for testing, which should reproduce the issue

```
import time
from machine import UART, deepsleep, LED, WDT, reset_cause, reset
import machine
import network
import sensor

TIMEOUT = 10
uart = UART(1, 115200, timeout_char=200)

#camera stuff
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.VGA)
sensor.set_auto_gain(False)
sensor.set_framebuffers(3)

#LEDS used for camera indication
led_ok = LED("LED_GREEN")
led_ok.off()
led_cam_on = LED("LED_BLUE")
led_cam_on.on()

uart.write(f'Start Scan at {time.time()}\r\n')
done = False
start = time.time()

try:
    sensor.skip_frames(time=100)
    while not done and (start + TIMEOUT >= time.time()):
        led_cam_on.toggle()
        led_ok.off()
        img = sensor.snapshot()
        img.lens_corr(1.8)
        for code in img.find_qrcodes():
            led_cam_on.off()
            led_ok.on()
            time.sleep_ms(250)
            led_ok.off()
            uart.write(f"Payload = {code.payload()}\r\n")
            done = True

    led_cam_on.off()
    led_ok.off()
    sensor.shutdown(True)

    uart.write(f'Going to sleep at {time.time()}\r\n')
    while not uart.txdone():
         time.sleep_ms(50)

    deepsleep(20000)

except Exception as e:
    uart.write(f"Exception = {e} at {time.time()}\r\n")

```

iabdalkader, 2024-06-05:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/issue-with-imageio-write-py-example-on-rt1062-stucks-writing-to-sd-card/9662/2

kdarras, 2025-09-17:
I obtained the same error with my script when running `sensor.skip_frames(time=2000)` on the RT1062 (4.7.0).
Trying to set the frame buffer to contain 3 frames before the frame skipping solved the problem, even though get_framebuffers() told me that it was still just 1...

kdarras, 2025-09-18:
Just installed the latest dev release, but running into another problem:

[image]
Is this command obsolete now, possibly?

kwagyeman, 2025-09-18:
The frame buffer architecture was changed around, and buffers have slightly different memory allocation sizes. That shouldn't fail unless you were pushing the system such that RAM was extremely limited. The previous code was more forgiving and would have allowed any buffer requested, shurnk buffers such that they all fit given the count, and then cropped images to fit the smaller buffers.

kwagyeman, 2026-09-29:
Re-tested on an RT1062 (OV5640) with current master (2026-09-29 build).

- 60 deep-sleep cycles of the script above (VGA RGB565, 3 buffers, QR scan 5 s, `deepsleep`): 0 timeouts.
- 99 cycles of `sensor.reset()` + `skip_frames(time=2000)` + snapshot with default buffers: 0 timeouts.
- 20 min continuous VGA capture + `find_blobs`, default buffers: 23,985 frames, 0 exceptions.

Default framebuffer count is now 3 on the RT1062, so the manual `set_framebuffers(3)` workaround is no longer needed. The CSI rework that offloads frame capture to hardware landed in the dev releases from Sept 2025 onward.

Closing. If anyone still hits `Frame capture timeout` on 4.8.x or later, open a new issue with the script and firmware version.

Notes from Claude: the three runs were driven from the HIL bench host over USB; deep-sleep wakes were verified by the CDC re-enumerating each cycle.

closed 2026-09-29 by kwagyeman