openmv/openmv #2202: A simple way to fix latest version of GC2145

cyhcyhgo, 2024-04-19:
Hello! After a few days of trying, I finally found the source of the problem and a solution to it.
1. The GC2145 that is used on ArduCam and Arduion Nicla Vision has been updated. The I2C slave address has been changed from 0x78 to 0x3C (I didn't find the latest version of sensor datasheet, but I saw this in Arduino dvp library in fig. 1), hence OV5640 will not conflict to GC2145.
[image]

2. On Arduino GIGA, PA1 is used to turn on the linear regulated power supply, the sensor will not work if it is not set high. This pin was defined as OMV_CSI_RESET_PIN, bu was commented out. 
[image]

Here is my solution, and it works on my Arduino GIGA(maybe also works on Arduino Nicla Vision):
1. Change the slave I2C address to 0x3C;
2. Modify sensor_utils.c, to check GC2145 firstly ( ince it has lower address than OV2640 0x60);
[image]

3. Enable OMV_CSI_RESET_PIN.

kwagyeman, 2024-04-19:
Hi, you should send a PR. Not an issue if you've resolved this. Please bundle up your changes and make a PR.

Thanks!

iabdalkader, 2024-04-19:
The I2C address hasn't changed, it can't change there's no pin to change the address. The I2C scanner returns the address right-shifted, so the addresses you see in `sensor.h` are the 8-bits addresses, so 0x3C << 1 is the same as 0x78.
It might be the disabled PA1 pin like you said, but it was disabled for a reason, maybe it doesn't work with the other sensors OV7670, OV7675. I'll try to re-enable it, if the other sensors still work I'll leave it enabled.

iabdalkader, 2024-04-19:
I enabled PA1, and tested OV7670 and OV7675 they both still work. Can you try the dev firmware and see if the GC2145 works now ?

cyhcyhgo, 2024-04-19:
> I enabled PA1, and tested OV7670 and OV7675 they both still work. Can you try the dev firmware and see if the GC2145 works now ?

Thanks for your reply. I have tested the latest version of development firmware, it works fine with GC2145.
On the old version of OV7675 board (only has one fpc connector, while the new version has three), there is a pull-up resistor(R3) to pull PA1 high, so you don't need to set this pin in firmware. But in the new version, this resistor has been removed.
[image]
[image]

iabdalkader, 2024-04-19:
> On the old version of OV7675 board (only has one fpc connector, while the new version has three), there is a pull-up resistor(R3) to pull PA1 high, so you don't need to set this pin in firmware. But in the new version, this resistor has been removed.

Ah that explains why it stopped working, but you mean GC2145 right ?

Anyway, seems this issue is now fixed.

cyhcyhgo, 2024-04-19:
> Ah that explains why it stopped working, but you mean GC2145 right ?
Yes, without pulling PA1 up (by defaut it is floating), the LDO will not work, hence GC2145 will not answer to i2c.
Btw, could you please add 
#define OMV_GC2145_ROTATE                     (1) 
after 
#define OMV_GC2145_ENABLE                     (1) 
in omv_boardconfig.h of Arduino GIGA ? The image output is upside down.

iabdalkader, 2024-04-19:
I see, well all 3 sensors should now be working.

Note I enabled the rotation.

closed 2024-04-19 by iabdalkader