openmv/openmv #2216: Corrupt Lines

kwagyeman, 2024-05-17:
When the number of frame buffers is set to 1 on the RT1062 you'll get those corrupt cache invalidation lines on small resolutions like QQVGA.

It doesn't seem to bother multiple buffer mode, which everyone uses. But, something to eventually fix.

iabdalkader, 2024-05-18:
Yeah I noticed this while testing `py_image_to`. It's very easy to reproduce if you convert the snapshot to anything, to forcing the lines to changes every frame. For example:

```Python
import sensor
import time
import image

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QQVGA)

while True:
    sensor.snapshot().to_rainbow(color_palette=image.PALETTE_IRONBOW)
```

> When the number of frame buffers is set to 1 
> That said, unlike before, it's only one line near the top left of the frame. Not all over it.
> It doesn't seem to bother multiple buffer mode, which everyone uses. But, something to eventually fix.

That's not accurate. I see multiple lines get corrupted, almost every line randomly, there's no obvious pattern. Also, unless `image_to` switches to 1 buffer, this happens with multiple buffers too.
[image] [image]

kwagyeman, 2024-05-20:
Okay, will spend time thinking more about this on Monday/Tuesday.

PR #1610 (2024-05-21): imlib: Invalidate vbuffers before returning them.
The CPU might fetch lines, speculatively, from a vbuffer after an invalidate and before the [M/E]DMA is done writing to it. In this case, when the DMA is done, the CPU will have some "outdated" lines in cache. Speculative access in CM7 is documented here https://developer.arm.com/documentation/ddi0489/f/memory-system/speculative-accesses

> Speculative cache linefills can be initiated to any Cacheable memory address, and in rare cases, regardless of whether there is any instruction that causes the cache linefill.

This change invalidates the bu [... cut]
commit 80fce781 (2024-05-21): imlib: Invalidate vbuffers before returning them.

* Fixes #2216
closed 2024-05-21 by iabdalkader