openmv/openmv #2217: Crash on jpeg decompress on H7/M7/M4.

kwagyeman, 2024-05-17:
In place or copy_to_fb jpeg decompression on the H7/M7/M4 crashes the camera. The H7 Plus and RT1062 don't have any issues. So, it's not an issue with code paths but something to do with memory exhaustion. Most likely, a memory overflow error is not handled correctly somewhere. I'm guessing the code should have thrown an error instead of crashing the camera.

This is blocking me from collecting performance test results, so I will fix it ASAP.

kwagyeman, 2024-07-26:
@bitbank2

kwagyeman, 2024-07-26:
```
import sensor, image, time

sensor.reset()
sensor.set_pixformat(sensor.GRAYSCALE)
sensor.set_framesize(sensor.QVGA)
sensor.set_framebuffers(1)

count = 0
avg = 0

while(True):
    img = sensor.snapshot().to_jpeg()
    start = time.ticks_us()

    img.to_grayscale() # Algorithm to test

    avg += time.ticks_diff(time.ticks_us(), start)
    count += 1
    print("%0.2f" % ((avg / count) / 1000))
```

bitbank2, 2024-07-29:
I don't own the M4/M7. I have the H7, H7P and RT1060. Is there any way to either allow me to recreate on the hardware I own or point me to where you think it is in the code?

kwagyeman, 2024-07-29:
@bitbank2 I'll try to narrow it down and see where the issue is. I should be able to create it on the RT1062 if it's your code.

kwagyeman, 2025-10-16:
@iabdalkader - Will check this when I get home. can't verify that this ix fixed until then since lacking M4/M7 on trip.