openmv/openmv #2240: Light shield on the RT1062 using pyb

EvanBartel, 2024-06-24:
For the RT1062 built-in examples, the light shield example is using pyb which is not supported by the RT1062.

**To Reproduce**
Steps to reproduce the beh
avior:
1. Connect the RT1062 board.
2. Click File->Examples->Light Shield->Light.py
3. Run script.
4. See error "ImportError: no module named 'pyb'"

The expected behavior is for the script to not use pyb and *possibly* use machine instead.

Script modifications will be needed to support the change.

 - IDE version: 4.1.5
 - Firmware version: 4.5.6
 - Board: RT1062
 - OS: Windows

EvanBartel, 2024-06-24:
Code link in the repo: https://github.com/openmv/openmv/blob/60df3e61fce74715be785a4b41677b630f1d4784/scripts/examples/50-OpenMV-Boards/60-Shields/68-Light-Shield/light.py

kwagyeman, 2024-06-24:
Yep, that's an automated response. Will fix this.

iabdalkader, 2024-06-24:
Yeah it's automated, need to switch that to bot.