openmv/openmv #2261: Increase frame transfer speed

kwagyeman, 2024-07-08:
In high speed USB the host has to poll us often which slows down performance:

[image]

[image]

Should look into how to reduce the need for polling an be ready.

kwagyeman, 2024-08-05:
This has been solved. Just need to verify the fix is safe on mac/linux/windows: https://github.com/hathach/tinyusb/discussions/2707

kwagyeman, 2024-08-05:
[image]
[image]
Using 4096 byte buffers there are no more NACKs.

Burst bandwidth is 25.65 MB/s (205.2 Mb/s) .

iabdalkader, 2024-08-05:
It could be improved once TinyUSB allows direct access to the FIFOs that's another memcpy saved.

kwagyeman, 2026-04-03:
This is fixed now.