openmv/openmv #2301: src/build/mimxrt.lds:109: non constant or forward reference address expression for section .gc.blocks.table

IAmCoder, 2024-07-20:
I get the following error when building:

> src/build/mimxrt.lds:109: non constant or forward reference address expression for section .gc.blocks.table
collect2: error: ld returned 1 exit status
make: *** [/src/omv/ports/mimxrt/omv_portconfig.mk:570: firmware] Error 1

I used to be able to build with my setup by running:
`make -j$(nproc) TARGET=OPENMV_RT1060 -C src`

kwagyeman, 2024-07-20:
You need to match the version of GCC we are using in tools/ci.sh

closed 2024-07-20 by kwagyeman

IAmCoder, 2024-07-20:
Thanks. I can build with that version of the toolchain now! I prepared a pull request in case the [README.md](https://github.com/openmv/openmv/blob/574fc72c352d28d6c8442a1d62bab5747675010c/src/README.md?plain=1#L46) needs to be updated: https://github.com/openmv/openmv/pull/2303.