openmv/openmv #2392: docker build failed

zhaoyuRobotics, 2024-08-28:
**Describe the bug**  

hello, i have a trobule when i want to compile this project by the docker. please tell me why and what shoule i do next . thanks

```
./.dapper -C ../ -f docker/Dockerfile --target dapper make
Sending build context to Docker daemon    610MB
Step 1/15 : FROM ubuntu:20.04 AS dapper
 ---> 9df6d6105df2
Step 2/15 : ARG DAPPER_HOST_ARCH
 ---> Using cache
 ---> 11a06a278daf
Step 3/15 : ENV DAPPER_ENV TARGET
 ---> Using cache
 ---> 8a34e572e4b0
Step 4/15 : ENV ARCH $DAPPER_HOST_ARCH
 ---> Using cache
 ---> 169244313d39
Step 5/15 : ENV DAPPER_OUTPUT ./docker/build
 ---> Using cache
 ---> d60568105b4c
Step 6/15 : ENV DAPPER_DOCKER_SOCKET true
 ---> Using cache
 ---> 0a30791d7d18
Step 7/15 : ENV DAPPER_TARGET dapper
 ---> Using cache
 ---> 6894a2eea2bb
Step 8/15 : RUN apt update && apt install -y build-essential wget git python3 python-is-python3
 ---> Using cache
 ---> 963d05d169bd
Step 9/15 : ENV GCC_URL="https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/10-2020q4/gcc-arm-none-eabi-10-2020-q4-major-x86_64-linux.tar.bz2"
 ---> Using cache
 ---> 94c543f6561a
Step 10/15 : RUN mkdir ./gcc
 ---> Using cache
 ---> 2e1e3cc28401
Step 11/15 : RUN wget --no-check-certificate -O - ${GCC_URL} | tar --strip-components=1 -jx -C ./gcc
 ---> Using cache
 ---> 19ce7c6556b5
Step 12/15 : RUN mkdir ./cmake &&     CMAKE_URL="https://cmake.org/files/v3.20/cmake-3.20.0-linux-x86_64.tar.gz" &&     wget --no-check-certificate -O - ${CMAKE_URL} | tar --strip-components=1 -xz -C ./cmake
 ---> Using cache
 ---> a8269e572139
Step 13/15 : RUN mkdir -p /source/gcc && mkdir -p /source/cmake && cp -r ./gcc /source/gcc && cp -r ./cmake /source/cmake
 ---> Using cache
 ---> f653d47bb691
Step 14/15 : WORKDIR /source
 ---> Using cache
 ---> 337afb158462
Step 15/15 : ENTRYPOINT ["./docker/build.sh"]
 ---> Using cache
 ---> c1ef34796a4f
Successfully built c1ef34796a4f
Successfully tagged openmv:HEAD
Sending build context to Docker daemon    610MB
Step 1/2 : FROM openmv:HEAD
 ---> c1ef34796a4f
Step 2/2 : COPY . /source/
 ---> 8b1251785139
Successfully built 8b1251785139
Successfully tagged openmv:HEAD
++ dirname ./docker/build.sh
+ cd ./docker/..
+ export PATH=/source/gcc/gcc/bin:/source/cmake/cmake/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ PATH=/source/gcc/gcc/bin:/source/cmake/cmake/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ git submodule update --init --depth=1
Submodule 'src/lib/tflm/libtflm' (https://github.com/openmv/libtflm.git) registered for path 'src/lib/tflm/libtflm'
Submodule 'src/micropython' (https://github.com/openmv/micropython.git) registered for path 'src/micropython'
Submodule 'src/omv/modules/ulab' (https://github.com/v923z/micropython-ulab.git) registered for path 'src/omv/modules/ulab'
Cloning into '/source/src/lib/tflm/libtflm'...
Cloning into '/source/src/micropython'...
[... 832 lines cut ...]
GEN ../../../build/lib/tflm/tflm_builtin_models.h
  File "/source/src/../tools/tflite2c.py", line 46
    print(f'VELA {model_name}.tflite\n{"\n".join(output)}\n', file=sys.stderr)
          ^
SyntaxError: f-string expression part cannot include a backslash
  File "/source/src/../tools/tflite2c.py", line 46
    print(f'VELA {model_name}.tflite\n{"\n".join(output)}\n', file=sys.stderr)
          ^
SyntaxError: f-string expression part cannot include a backslash
make[1]: *** [Makefile:61: /source/src/build/lib/tflm/tflm_builtin_models.c] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:56: /source/src/build/lib/tflm/tflm_builtin_models.h] Error 1
make[1]: Leaving directory '/source/src/lib/tflm'
make: *** [/source/src/omv/ports/stm32/omv_portconfig.mk:662: FIRMWARE_OBJS] Error 2
make: Leaving directory '/source/src'
FATA[0120] exit status 2                                
Makefile:12: recipe for target 'in-docker-build' failed
make: *** [in-docker-build] Error 1

```

**Please complete the following information**
 - linux:Linux ubuntu 5.4.0-150-generic #167~18.04.1-Ubuntu SMP Wed May 24 00:51:42 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
 - Docker version 20.10.21, build 20.10.21-0ubuntu1~18.04.3  
 - Python 3.6.9

zhaoyuRobotics, 2024-08-28:
branch is v4.5.8

iabdalkader, 2024-08-28:
Try building the master branch it should work, or use a newer Python.

zhaoyuRobotics, 2024-08-29:
thank you reply. but i think it uses the docker`s python , not my local python 
```
Step 8/15 : RUN apt update && apt install -y build-essential wget git python3 python-is-python3
 ---> Using cache
 ---> 963d05d169bd
```

i try master branch ， but still have trouble 
```
make: Entering directory '/source/src'
make[1]: Entering directory '/source/src/micropython/ports/stm32'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from /source/src/omv/modules
GEN /source/src/build/micropython/genhdr/mpversion.h
GEN /source/src/build/micropython/frozen_content.c
CC /source/src/build/micropython/frozen_content.c
make[1]: Leaving directory '/source/src/micropython/ports/stm32'
/source/gcc/gcc/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld:/source/src/build/stm32fxxx.lds:62: non constant or forward reference address expression for section .gc.blocks.table
collect2: error: ld returned 1 exit status
make: *** [/source/src/omv/ports/stm32/omv_portconfig.mk:697: firmware] Error 1
make: Leaving directory '/source/src'
FATA[0175] exit status 2                                
Makefile:12: recipe for target 'in-docker-build' failed
make: *** [in-docker-build] Error 1

```

maybe the gnu version does not support this syntax
```
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

```

sir，i want to know which ubuntu version can support smooth compilation> i am current using 18.04

zhaoyuRobotics, 2024-08-29:
log:  

```
EPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  633.8MB
Step 1/15 : FROM ubuntu:20.04 AS dapper
 ---> 9df6d6105df2
Step 2/15 : ARG DAPPER_HOST_ARCH
 ---> Using cache
 ---> 8eeb07a4c2d0
Step 3/15 : ENV DAPPER_ENV TARGET
 ---> Using cache
 ---> 955eb1422ad8
Step 4/15 : ENV ARCH $DAPPER_HOST_ARCH
 ---> Using cache
 ---> aa8f1c09ee91
Step 5/15 : ENV DAPPER_OUTPUT ./docker/build
 ---> Using cache
 ---> 147afcf80d1a
Step 6/15 : ENV DAPPER_DOCKER_SOCKET true
 ---> Using cache
 ---> f696591565e5
Step 7/15 : ENV DAPPER_TARGET dapper
 ---> Using cache
 ---> 13a772fbf863
Step 8/15 : RUN apt update && apt install -y build-essential wget git python3 python-is-python3
 ---> Using cache
 ---> fae37a75ee63
Step 9/15 : ENV GCC_URL="https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/10-2020q4/gcc-arm-none-eabi-10-2020-q4-major-x86_64-linux.tar.bz2"
 ---> Using cache
 ---> 094bb62f04d7
Step 10/15 : RUN mkdir ./gcc
 ---> Using cache
 ---> 857a034d9095
Step 11/15 : RUN wget --no-check-certificate -O - ${GCC_URL} | tar --strip-components=1 -jx -C ./gcc
 ---> Using cache
 ---> 6da45e7d07df
Step 12/15 : RUN mkdir ./cmake &&     CMAKE_URL="https://cmake.org/files/v3.20/cmake-3.20.0-linux-x86_64.tar.gz" &&     wget --no-check-certificate -O - ${CMAKE_URL} | tar --strip-components=1 -xz -C ./cmake
 ---> Using cache
 ---> d685bfa7db33
Step 13/15 : RUN mkdir -p /source/gcc && mkdir -p /source/cmake && cp -r ./gcc /source/gcc && cp -r ./cmake /source/cmake
 ---> Using cache
 ---> 1c6a91f69079
Step 14/15 : WORKDIR /source
 ---> Using cache
 ---> 1310069a7500
Step 15/15 : ENTRYPOINT ["./docker/build.sh"]
 ---> Using cache
 ---> 2a1112bfbba2
Successfully built 2a1112bfbba2
Successfully tagged openmv:master
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  633.8MB
Step 1/2 : FROM openmv:master
 ---> 2a1112bfbba2
Step 2/2 : COPY . /source/
 ---> daff4d3a869d
Successfully built daff4d3a869d
Successfully tagged openmv:master
++ dirname ./docker/build.sh
+ cd ./docker/..
+ export PATH=/source/gcc/gcc/bin:/source/cmake/cmake/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ PATH=/source/gcc/gcc/bin:/source/cmake/cmake/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ git submodule update --init --depth=1
[... 1060 lines cut ...]
CC src/usbd_ctlreq.c
CC src/usbd_ctlreq.c
CC src/stm32fxxx_it.c
CC src/stm32fxxx_it.c
CC src/string0.c
CC src/flash.c
CC src/string0.c
CC src/usbd_uvc.c
CC src/usbd_cdc.c
CC src/usbd_ioreq.c
CC src/usbd_ioreq.c
CC src/usbd_desc.c
CC src/usbd_desc.c
CC src/qspif.c
CC src/systick.c
make[1]: Leaving directory '/source/src/omv/ports/stm32/boot'
make[1]: Leaving directory '/source/src/omv/ports/stm32/uvc'
/source/gcc/gcc/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld:/source/src/build/stm32fxxx.lds:63: syntax error
collect2: error: ld returned 1 exit status
make: *** [/source/src/omv/ports/stm32/omv_portconfig.mk:697: firmware] Error 1
make: Leaving directory '/source/src'
FATA[0199] exit status 2                                
make: *** [Makefile:12: in-docker-build] Error 1

```

iabdalkader, 2024-08-29:
If you pull the master again now it should be fixed.

zhaoyuRobotics, 2024-08-29:
Thankd for you reply, i test it . but it is stuck here:

```
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  634.4MB
Step 1/19 : FROM ubuntu:20.04 AS dapper
 ---> 9df6d6105df2
Step 2/19 : ARG DAPPER_HOST_ARCH
 ---> Using cache
 ---> 8eeb07a4c2d0
Step 3/19 : ENV DAPPER_ENV TARGET
 ---> Using cache
 ---> 955eb1422ad8
Step 4/19 : ENV ARCH $DAPPER_HOST_ARCH
 ---> Using cache
 ---> aa8f1c09ee91
Step 5/19 : ENV DAPPER_OUTPUT ./docker/build
 ---> Using cache
 ---> 147afcf80d1a
Step 6/19 : ENV DAPPER_DOCKER_SOCKET true
 ---> Using cache
 ---> f696591565e5
Step 7/19 : ENV DAPPER_TARGET dapper
 ---> Using cache
 ---> 13a772fbf863
Step 8/19 : RUN apt update && apt install -y build-essential wget git python3 python-is-python3
 ---> Using cache
 ---> fae37a75ee63
Step 9/19 : RUN mkdir -p /source/gcc
 ---> Running in 2f4c872d9a52
Removing intermediate container 2f4c872d9a52
 ---> 40e59ad9df5a
Step 10/19 : ENV GCC_URL="https://developer.arm.com/-/media/Files/downloads/gnu/13.2.rel1/binrel/arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eabi.tar.xz"
 ---> Running in 928702b071d4
Removing intermediate container 928702b071d4
 ---> 89b79f15b5e4
Step 11/19 : RUN wget --no-check-certificate -O - ${GCC_URL} | tar --strip-components=1 -Jx -C /source/gcc
 ---> Running in 62eac72a8f76
--2024-08-29 15:28:51--  https://developer.arm.com/-/media/Files/downloads/gnu/13.2.rel1/binrel/arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eabi.tar.xz
Resolving developer.arm.com (developer.arm.com)... 23.210.215.120, 23.199.34.121
Connecting to developer.arm.com (developer.arm.com)|23.210.215.120|:443... connected.

```

iabdalkader, 2024-08-29:
That's a connection issue I can't fix.

zhaoyuRobotics, 2024-08-31:
But my cpmputer can download it locally, i suspect it is aproblem with Docker‘s proxy.

```
 $ wget --no-check-certificate -O - ${GCC_URL} | tar --strip-components=1 -Jx -C ./build

--2024-08-30 21:02:08--  https://developer.arm.com/-/media/Files/downloads/gnu/13.2.rel1/binrel/arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eabi.tar.xz
Connecting to 192.168.56.1:7890... connected.
Proxy request sent, awaiting response... 302 Moved Temporarily
Location: https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu/13.2.rel1/binrel/arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eabi.tar.xz [following]
--2024-08-30 21:02:08--  https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu/13.2.rel1/binrel/arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eabi.tar.xz
Connecting to 192.168.56.1:7890... connected.
Proxy request sent, awaiting response... 200 OK
Length: 179347712 (171M) [application/octet-stream]
Saving to: ‘STDOUT’

-                                                  100%[===============================================================================================================>] 171.04M   616KB/s    in 5m 51s  

2024-08-30 21:08:01 (499 KB/s) - written to stdout [179347712/179347712]

```

zhaoyuRobotics, 2024-08-31:
finally , I successfully compiled the firmware on my machine。  

```
sudo apt install gcc-arm-none-eabi 
cd cd micropython/mpy-cross/
make 
cd ../..
make TARGET=OPENMV4
```

**uname -a** : inux zy 6.8.0-40-generic #40~22.04.3-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 30 17:30:19 UTC 2 x86_64 x86_64 x86_64 GNU/Linux  

**arm-none-eabi-gcc --version**
arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

**arm-none-eabi-g++ --version**
arm-none-eabi-g++ (15:10.3-2021.07-4) 10.3.1 20210621 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

closed 2024-08-31 by zhaoyuRobotics