openmv/openmv #2469: Possible memory corruption/allocation error with image.Image(ndarray)

Jarvis-X, 2024-10-30:
**Describe the bug**
As described in [this post](https://forums.openmv.io/t/memory-corruption-with-image-image-ndarray/10146). 

**To Reproduce**
- The full program (first the desired behavior, then change one line of code to reproduce the errors)
  - Download and execute [this program](https://github.com/LehighBlimpGroup/Blob-detection-and-Tracking/blob/main/perception_subsystem.py) on either a NiclaVision (recommended for reproducing the errors sooner and more easily) or an OpenMV Cam RT1062. 
  - You might need to tweak the RGB gains on [line 49 to 52](https://github.com/LehighBlimpGroup/Blob-detection-and-Tracking/blob/main/perception_subsystem.py#L49-L52) a bit so that the image looks more naturally colored. 
    - To do that, on OpenMV Cam RT1062, point the camera in the environment and run the program, then copy the lone 3-element tuple from the console to [line 50](https://github.com/LehighBlimpGroup/Blob-detection-and-Tracking/blob/main/perception_subsystem.py#L50)
    - On NiclaVision, run [this program](https://github.com/LehighBlimpGroup/Blob-detection-and-Tracking/blob/main/get_gains.py) once and copy the values in the console after "RGB gain:" to [line 52](https://github.com/LehighBlimpGroup/Blob-detection-and-Tracking/blob/main/perception_subsystem.py#L52)
  - Point the camera towards a color wheel where purple or green or red colors are presented until cells of the corresponding colors show up in the image buffer. The LED should change colors/blink/dim. Keep the program running for as long as you want. It should not crash or throw any error.
  - *So far, we consider the behaviors "stable" as it is NOT using image.Image(ndarray) methods that I am suspecting to be buggy.*
  - Now, go to [line 449](https://github.com/LehighBlimpGroup/Blob-detection-and-Tracking/blob/main/perception_subsystem.py#L449) and replace `_, _, max_val, same_val_rc = self.optimized_loop(metric)` with `_, _, max_val, same_val_rc = self.kernel_count(metric)`
  - Rerun the program for ~2 mins max. We should see the program crash with the errors mentioned in the original post.
  #### Basically, both `optimized_loop(metric)` and `kernel_count(metric)` convolve the `metric` array that we treat as a (N_ROWS, N_COLS) matrix with the static `kernel_matrix`. The difference is that `optimized_loop()` does that with nested loops, and `kernel_count(metric)` does that by converting the `metric` into a grayscale image first, and applying `morph(kernel_matrix)` on it, then converting the image back to an ndarray. To further narrow the scope of the problem, keep our change on `kernel_count(metric)`, and set `FILTER_KERNEL = True` on [line 127](https://github.com/LehighBlimpGroup/Blob-detection-and-Tracking/blob/main/perception_subsystem.py#L127), which skips the `image.Image(ndarray_metric, buffer=self.grid_img_bf)` and `morph()`. The program should again run forever.
  

- The following code snippet serves as a minimum replicable example for my suspicion on the `image.Image(ndarray)` method but does not really give the same error as in my full program. Run it with a NiclaVision for at most 1 min to see the memory allocation error.
``` python
import sensor
import time
import micropython
import array
from machine import I2C
from vl53l1x import VL53L1X
import image
import math
from machine import Pin
import omv
import random
import asyncio
from ulab import numpy as np

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time = 2000)

clock = time.clock()

# Grid setup
N_ROWS = 160
N_COLS = 160
KERNEL_SIZE = 5 # a n x n kernel that summarizes the possibility scores inside to give the most
                # approachable detection, for target balloon detections only
KERNEL_CENTER_EMPH = 3 # how much weight we want the center of the kernel to count
kernel_scale = KERNEL_SIZE ** 2 + KERNEL_CENTER_EMPH - 1
# setting up the kernel matrix
kernel_matrix = []
for i in range(KERNEL_SIZE):
    kernel_row = []
    for j in range(KERNEL_SIZE):
        if i == KERNEL_SIZE // 2 and j == KERNEL_SIZE // 2:
            kernel_row.append(KERNEL_CENTER_EMPH)
        else:
            kernel_row.append(1)
    kernel_matrix.append(kernel_row)

class ManyMembers:
    def __init__(self):
        self.dictionay_of_things = {float(random.random()): f"value_{i}" for i in range(128)}
        self.some_float = random.random()*100.0
        self.a_list = list(range(100))
        self.np_rng = np.random.Generator(123456)

    def kernel_count(self, kernel_matrix):
        input_matrix = self.np_rng.random(size=(N_ROWS, N_COLS))*256.0/kernel_scale
        metric_image = image.Image(input_matrix)
        metric_image.morph(KERNEL_SIZE//2, kernel_matrix)
        ndarray_morphed_metric = metric_image.to_ndarray(dtype=np.uint8).flatten()
        max_val = np.max(ndarray_morphed_metric)
        return max_val

if __name__ == "__main__":
    many_member_class = ManyMembers()
    while(True):
        clock.tick()
        img = sensor.snapshot()
        print(many_member_class.kernel_count(kernel_matrix))
        print(clock.fps())
```

**Expected behavior**
As stated above. Sorry if the code and descriptions are lengthy. It is quite challenging for me to even locate the issue in the beggining.

**Please complete the following information**
 - IDE version: 4.2.4
 - Firmware version: NicalVision 4.5.9; OpenMV Cam RT1062 4.6.0
 - Board: NicalVision and OpenMV Cam RT1062
 - OS: Windows 11

iabdalkader, 2024-10-30:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/memory-corruption-with-image-image-ndarray/10146/4

Jarvis-X, 2024-10-31:
Note that the program https://github.com/LehighBlimpGroup/Blob-detection-and-Tracking/blob/main/perception_subsystem.py has been updated. To reproduce the errors, go to the history version of it at https://github.com/LehighBlimpGroup/Blob-detection-and-Tracking/blob/5c0717a0d25804d4b597cb93e233405422b7a37b/perception_subsystem.py

kwagyeman, 2024-10-31:
Thanks for the update.

kwagyeman, 2024-11-07:
@Jarvis-X, can you test this Nicla firmware and let me know if your issue is fixed? It has the change below. I've run your demo program and followed your steps but it doesn't crash. I did manage to see the Nicla out of memory error given the other test script example. However, to verify with floats I had to reduce the allocation size from 160x160 to 160x140. After doing that floats/uint8 didn't crash anymore. So, it's not clear if the issue was just heap fragmentation.

[firmware.zip](https://github.com/user-attachments/files/17656952/firmware.zip)

I made the change here on this PR: https://github.com/openmv/openmv/pull/2487

kwagyeman, 2024-11-07:
Note that Image.Image() converting from an ndarray is unlikely to corrupt memory. It is coded in a way where it won't write outside of the image. That said, an ndarray with invalid dimensions could cause it to do the wrong thing.

closed 2024-11-22 by iabdalkader