openmv/openmv #2474: There is an error when compiling the firmware for Raspberry Pi Pico, but compiling ARDUINO_NANO_RP2040_CONNECT is possible.

yueritian, 2024-11-01:
During compilation, an error is initially reported that PICO is not found. After modifying it to RPI_PICO and recompiling, the following error occurs:
Install the project...
-- Install configuration: "Release"
make[5]: Leaving directory '/Source/openmv/openmv/src/build/rp2/_deps/picotool-build'
[  3%] Performing test step for 'picotoolBuild'
picotool v2.0.0 (Linux, GNU-13.2.0, Release)
[  3%] Completed 'picotoolBuild'
make[4]: Leaving directory '/Source/openmv/openmv/src/build/rp2'
[  3%] Built target picotoolBuild
make[3]: Leaving directory '/Source/openmv/openmv/src/build/rp2'
make[2]: *** [Makefile:91: all] Error 2
make[2]: Leaving directory '/Source/openmv/openmv/src/build/rp2'
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make[1]: *** [Makefile:59: all] Error 1
make[1]: Leaving directory '/Source/openmv/openmv/src/micropython/ports/rp2'
make: *** [/Source/openmv/openmv/src/omv/ports/rp2/omv_portconfig.mk:37: firmware] Error 2
make: Leaving directory '/Source/openmv/openmv/src'

Could you please tell me how I should troubleshoot the error?

=============================
Here is the screenshot of my successful compilation of ARDUINO_NANO_RP2040_CONNECT:[image]

kwagyeman, 2024-11-01:
Please duplicate the build process that github actions does: https://github.com/openmv/openmv/blob/master/.github/workflows/firmware.yml

closed 2024-11-01 by kwagyeman

iabdalkader, 2024-11-03:
The Pico just doesn't build anymore. I have no time to find out why or to fix it, it's a very useless MCU anyway.

yueritian, 2024-11-04:
Got it, thanks a lot. No biggie, I'm just trying to learn as much as I can from this whole thing. Could you maybe throw some advice my way? I've been running the build-firmware workflow on my machine using ACT, and it's been smooth sailing with ARDUINO_NANO_RP2040_CONNECT. But I'm having trouble with the PICO; I've tried making some changes, but it's still not working out. when generating genhdr/qstr.i.last：| <command-line>: error: empty filename in #include。

> The Pico just doesn't build anymore. I have no time to find out why or to fix it, it's a very useless MCU anyway.

iabdalkader, 2024-11-04:
> Could you maybe throw some advice my way

It seems like one of the headers defined on the command line is missing (for example, in Makefiles headers are defined like `CMSIS_H='<rp2.h>'`).

yueritian, 2024-11-05:
Thanks！I've compile it successfully, though it might not be super stable. 
**omv/boards/** 
1.PICO - > RPI_PICO
2.omv_boardconfig.cmake | add:RP2040.h
3.add files:main.py，readme_txt.py (copy from ARDUINO_NANO_RP2040_CONNECT，led 6 - > led 25)
4.imlib_config.h | add：IMLIB_ENABLE_IMAGE_FILE_IO
5.omv_boardconfig.h | add:OMV_TUSBDBG_ENABLE、OMV_TUSBDBG_BUFFER

**micropython/ports/rp2/boards/RPI_PICO**
1.mpconfigboard.h | add：MICROPY_HW_USB_MSC

That's all.

iabdalkader, 2024-11-05:
Great! 

>  1.PICO - > RPI_PICO

I already renamed this in the repo a day or two ago. I noticed they renamed the board.

> micropython/ports/rp2/boards/RPI_PICO
> 1.mpconfigboard.h | add：MICROPY_HW_USB_MSC

You can probably define this in `omv/ports/rp2/omv_mpconfigport.h` if not defined.

You're welcome to send a PR to fix the build if you like.

yueritian, 2024-11-05:
> You're welcome to send a PR to fix the build if you like.

Copy that, the firmware has been compiled successfully, but I'm unable to connect it with the IDE; I'm still looking into the cause.

iabdalkader, 2024-11-05:
> I'm unable to connect it with the IDE;

The IDE will not recognize the PICO USB ID (CC @kwagyeman if you want to add it)

I sent a fix to the build, it should build now, but it's untested and unknown if it works.

kwagyeman, 2024-11-05:
@yueritian - The IDE will allow connecting to this:

```
{
            "hidden" : false,
            "boardType" : "PICO",
            "boardArchString" : "PICO M0",
            "boardDisplayName" : "Arduino Nano RP2040 Connect (RP2040)",
            "boardFirmwareFolder" : "ARDUINO_NANO_RP2040_CONNECT",
            "boardVidPid" : "2341:005e",
            "boardPidMask" : "0x00FF",
            "boardSerialNumberInverseFilters" : [],
            "bootloaderVidPid" : "28ea:0003",
            "bootloaderType" : "picotool",
            "bootloaderSettings" : {
                "altvidpidDisplayName" : "Arduino Nano RP2040 Connect (RP2040) Legacy",
                "altvidpid" : "2341:805e",
                "eraseCommands" : ["-o 0x10800000 -t bin"],
                "binProgramCommand" : "-x"
            }
        },
        {
            "hidden" : true,
            "boardType" : "PICO",
            "boardArchString" : "PICO M0",
            "boardDisplayName" : "Arduino Nano RP2040 Connect (RP2040) UF2",
            "boardFirmwareFolder" : "ARDUINO_NANO_RP2040_CONNECT",
            "boardVidPid" : "28ea:0003",
            "boardPidMask" : "0xFFFF",
            "boardSerialNumberInverseFilters" : [],
            "bootloaderVidPid" : "0000:0000",
            "bootloaderType" : "picotool",
            "bootloaderSettings" : {
                "altvidpidDisplayName" : "Arduino Nano RP2040 Connect (RP2040) Legacy",
                "altvidpid" : "2341:805e",
                "eraseCommands" : ["-o 0x10800000 -t bin"],
                "binProgramCommand" : "-x"
            }
        }
```

If the board VID:PID is 28ea:0003 then the IDE will connect to it. However, I haven't necessarily tested this. Everything works smoothly with the "Arduino Nano RP2040 Connect (RP2040)".

yueritian, 2024-11-06:
I have made some settings in micropython/ports/rp2/boards/RPI_PICO/mpconfigboard.h, like this:
```
#define MICROPY_HW_USB_MSC                                          (1)
#define MICROPY_HW_USB_VID                                            (0x28ea)
#define MICROPY_HW_USB_PID                                            (0x0003)
#define MICROPY_HW_USB_CDC_1200BPS_TOUCH		(1)
```
I have located the setting.json and made the configurations like this:
```
{
            "hidden" : false,
            "boardType" : "PICO",
            "boardArchString" : "PICO M0",
            "boardDisplayName" : "Raspberry Pi Pico",
            "boardFirmwareFolder" : "RPI_PICO",
            "boardVidPid" : "2341:005e",
            "boardPidMask" : "0x00FF",
            "boardSerialNumberInverseFilters" : [],
            "bootloaderVidPid" : "28ea:0003",
            "bootloaderType" : "picotool",
            "bootloaderSettings" : {
                "altvidpidDisplayName" : "Raspberry Pi Pico",
                "altvidpid" : "2341:805e",
                "eraseCommands" : ["-o 0x10800000 -t bin"],
                "binProgramCommand" : "-x"
            }
        },
        {
            "hidden" : true,
            "boardType" : "PICO",
            "boardArchString" : "PICO M0",
            "boardDisplayName" : "Raspberry Pi Pico UF2",
            "boardFirmwareFolder" : "RPI_PICO",
            "boardVidPid" : "28ea:0003",
            "boardPidMask" : "0xFFFF",
            "boardSerialNumberInverseFilters" : [],
            "bootloaderVidPid" : "0000:0000",
            "bootloaderType" : "picotool",
            "bootloaderSettings" : {
                "altvidpidDisplayName" : "Raspberry Pi Pico",
                "altvidpid" : "2341:805e",
                "eraseCommands" : ["-o 0x10800000 -t bin"],
                "binProgramCommand" : "-x"
            }
        }
```
It still doesn't work, and I'm not sure whether it's an issue with the firmware or the configuration.
Are there any methods to verify the debug firmware? OpenOCD+GDB?
ps：
I noticed in micropython/ports/rp2/mpconfigport.h, the VID is 0x2E8A, and the PID is 0x0005.
I saw in micropython/ports/rp2/boards/ARDUINO_NANO_RP2040_CONNECT/mpconfigboard.h, the VID is 0x2341, and the PID is 0x015e.So I don't quite understand how the matching was achieved.

kwagyeman, 2024-11-06:
The IDE will connect to anyboard with VID = boardVid and PID = boardPid & boardPidMask. Bootloader support is more complex. But, you should verify that you see a CDC USB port appear with those settings.

yueritian, 2024-11-06:
Yes, I've also noticed that there's no response with the above configuration. If I flash the firmware for 'Arduino Nano RP2040 Connect (RP2040)', although it doesn't work properly, there is a USB response.

kwagyeman, 2024-11-06:
Yes, but, does a USB CDC device appear in your windows device manager, linux /dev, or etc? If that doesn't appear then USB isn't up.

yueritian, 2024-11-07:
Things are getting a bit strange.

I cloned OpenMV again and generated the firmware for **ARDUINO_NANO_RP2040_CONNECT**（using Docker）. The generated firmware is 2KB larger than the one in the IDE(bin), and after flashing the firmware, there is no USB CDC device appear.

After flashing the UF2 firmware from the IDE, the USB will appear.

yueritian, 2024-11-07:
I downloaded versions 4.5.6, 4.5.7, 4.5.8, 4.5.9, and the Development Release from the OpenMV GitHub releases page. Version 4.5.6 is the one that comes with the new IDE. I flashed these firmware versions onto my RPI_PICO one by one, and versions 4.5.6, 4.5.7, 4.5.8, and 4.5.9 are recognized via USB, but **the Development Release is not**.

yueritian, 2024-11-08:
I cloned MicroPython and generated the firmware, but it couldn't be used properly. I switched to version v1.24.0, regenerated the firmware, and it worked fine. I added MICROPY_HW_USB_MSC, regenerated the firmware, and it was also usable, resulting in the disk being directly accessible. I think I should wait for OpenMV to release version 4.5.10 before continuing to look into this issue.  See you：D

iabdalkader, 2024-11-12:
The firmware should be working now, I just tested it.

yueritian, 2024-11-12:
That's great !  the IDE can now display information. 
[image]

However, I still get the message **'This computer cannot read the disk you connected.'** (I'm using a Mac) I think it's still a USB issue (Maybe it's something with TinyUSB), but my knowledge is somewhat limited. Could you give me some debugging advice?

iabdalkader, 2024-11-12:
Maybe you need to erase the filesystem, but the command for this is wrong in the IDE. I think it needs to be
```
"eraseCommands" : ["-o 0x10200000 -t bin"],
```

PR #2496 (2024-11-12): ports/rp2: Fix firmware.
Fixes  #2474