openmv/openmv #2507: The problem with find_rect in finding the largest rectangle

Gbone3176, 2024-11-22:
The find_rect function fails to find the largest rectangle when rotating the checkerboard by 45° clockwise, but it does find the largest rectangle when rotating it by 45° counterclockwise.

My code is as follows:
···
import sensor, image, time

# 初始化摄像头sensor.reset()
sensor.set_pixformat(sensor.RGB565)  # grayscale is faster (160x120 max on OpenMV-M7)
sensor.set_framesize(sensor.QQVGA)
sensor.set_vflip(True)
sensor.set_hmirror(True)
#sensor.skip_frames(time=2000)
sensor.reset()
sensor.set_pixformat(sensor.RGB565)  # grayscale is faster (160x120 max on OpenMV-M7)
sensor.set_framesize(sensor.QQVGA)
sensor.set_vflip(True)
sensor.set_hmirror(True)
sensor.skip_frames(time=2000)
# 开始帧采集
clock = time.clock()

while(True):
    clock.tick()
    img = sensor.snapshot()          # 获取一帧图像

    # 将图像二值化
    # 这里根据图像亮度手动调整阈值，适当设置会提高效果
    img.binary([(40, 255)])          # 二值化，选择一个合理的阈值

    # 寻找矩形
    rects = img.find_rects(threshold=8000)  # 阈值参数用于控制检测灵敏度，可调整

    # 绘制检测到的矩形
    for r in rects:
        img.draw_rectangle(r.rect(), color=(255, 0, 0))   # 绘制矩形边框
        print(r.w())
···

Rotate the hour hand clockwise:

[image]

To rotate counterclockwise:

[image]

kwagyeman, 2024-11-22:
Hi, what's the behavior if you don't binarize the image first?

Gbone3176, 2024-11-22:
> Hi, what's the behavior if you don't binarize the image first?

Hi ! I don't have any images without binarization, but the problem remains the same despite not using binarization for preprocessing ┭┮﹏┭┮

eiobo, 2025-12-07:
Any solution ？