openmv/openmv #2551: LCD Not Working When Running Offline with Latest Firmware

Salieri-Coin, 2024-12-19:
Describe the bug

When debugging with the OpenMV IDE, the screen displays normally. However, when the script is saved as main.py and run offline (without the IDE), the screen freezes or continuously displays overlapping/repeated frames.

To Reproduce

Steps to reproduce the behavior:

Write a script that works correctly when debugging in the OpenMV IDE.
Save the script as main.py.
Disconnect from the IDE and let the script run offline.
Observe that the screen either freezes or the display repeatedly overlays the same frames.
Expected behavior

The screen should display the expected content without freezing or overlapping images when running offline from the main.py script, just as it does when debugging with the IDE.
Please complete the following information

IDE version: 4.4.7
Firmware version: 4.6.0
Board: OpenMV Cam H7 Plus
OS: Windows

Salieri-Coin, 2024-12-20:
> Hi, can you post the code that has the issue?

import sensor
import display

sensor.reset()  # Initialize the camera sensor.
sensor.set_pixformat(sensor.RGB565)  # or sensor.GRAYSCALE
sensor.set_framesize(sensor.QQVGA2)  # Special 128x160 framesize for LCD Shield.
# Initialize the lcd screen.
# Note: A DAC or a PWM backlight controller can be used to control the
# backlight intensity if supported:
#  lcd = display.SPIDisplay(backlight=display.DACBacklight(channel=2))
#  lcd.backlight(25) # 25% intensity
# Otherwise the default GPIO (on/off) controller is used.
lcd = display.SPIDisplay()

while True:
    lcd.write(sensor.snapshot())  # Take a picture and display the image.

kwagyeman, 2024-12-20:
I just ran this on the H7 Plus using the v4.5.9 and it works.

I can confirm v4.6.0 has a bug.

closed 2024-12-31 by iabdalkader