openmv/openmv #2621: Waking machine.lightsleep(): timer and external interruption

snapchaturvedi, 2025-02-24:
**Describe the bug**
I am waking `machine.lightsleep()` in two different ways- using `ExtInt()` and using time delay, but one is overriding the other.

**To Reproduce**
Eg 1: `ExtInt()` waking, timer not working
```
import machine, pyb

# Callback for ExtInT
def cb():
    pass

# ExtInt()
pin = pyb.Pin("D6", pyb.Pin.IN, pyb.Pin.PULL_DOWN)
ext = pyb.ExtInt(pin, pyb.ExtInt.IRQ_FALLING, pyb.pin.PULL_DOWN, cb)

# Function called in main loop
def body():
    machine.lightsleep(10*60*1000)

# Main loop
while True:
    pyb.delay(5000)
    machine.lightsleep()
    body()
```

Eg 2: timer waking, `ExtInt()` not working
```
import machine, pyb

from pyb import LED

red_led = LED(1)
green_led = LED(2)
blue_led = LED(3)

# Callback for ExtInT
def cb():
    pass

# ExtInt()
pin = pyb.Pin("D6", pyb.Pin.IN, pyb.Pin.PULL_DOWN)
ext = pyb.ExtInt(pin, pyb.ExtInt.IRQ_FALLING, pyb.Pin.PULL_DOWN, cb)

# Function called in main loop
def body():
    machine.lightsleep(10*60*1000)

green_led.on()

# Main loop
while True:
    blue_led.on()
    pyb.delay(1000)
    machine.lightsleep()
    blue_led.off()
    pyb.delay(1000)
    red_led.on()
    pyb.delay(1000)
    red_led.off()
    machine.lightsleep(10000)
```

**Expected behavior**
`machine.lightsleep()` should sleep indefinetly until `ExtInt()` wakes it up, and then sleep for `xx` min before waking up.

**Please complete the following information**
 - IDE version: 
 - Firmware version: 4.5.9
 - Board: Portenta
 - OS: Windows

iabdalkader, 2025-02-24:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/lightsleep-v-deepsleep-on-portenta/10593/9

snapchaturvedi, 2025-02-24:
Hi @iabdalkader, that forum post is by me haha. Kwabena asked me to report the issue as he confirmed it's a bug

PC

iabdalkader, 2025-03-04:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/memory-allocation-failed-issue-on-portenta/10651/7

snapchaturvedi, 2025-03-04:
Hi @iabdalkader, sorry for the delay.

I tried the above 2 programs on Micropython v1.24.1 and am experiencing the same issue. I'll report this directly to MP, thanks!

PC

kwagyeman, 2025-03-05:
Issue here: https://github.com/micropython/micropython/issues/16862