openmv/openmv #2647: Unable to use pin D6/PA8 for External Interrupt

snapchaturvedi, 2025-04-04:
**Describe the bug**
I am using the following simple program based on `extint_wakeup.py` example to wakeup Portenta from lightsleep but as soon as I run the code it wakes up without any external interruption.

**To Reproduce**
Run the following program on Portenta H7 with Vision Shield (LoRa):
```py
import time
import pyb
import machine
from pyb import Pin, ExtInt

def callback(line):
    pass

led = pyb.LED(3)
pin = Pin("D6", Pin.IN, Pin.PULL_DOWN)
ext = ExtInt(pin, ExtInt.IRQ_RISING, Pin.PULL_DOWN, callback)

# Enter Stop Mode. Note the IDE will disconnect.
machine.sleep()

while True:
    led.on()
    time.sleep_ms(100)
    led.off()
    time.sleep_ms(100)
```

**Expected behavior**
The Portenta should stay in sleep mode until external voltage is provided to pin D6 for it to wake up.

**Please complete the following information**
 - IDE version: 4.2.4
 - Firmware version: 4.5.9
 - Board: Arduino Portenta + Vision Shield (LoRa)
 - OS: Windows

kwagyeman, 2025-04-04:
Hi, the lightsleep() function on the Portenta is bugged at the moment. Please avoid using it. Thank you for reporting the bug, though.  

If possible, could you please report the issue to Arduino?

snapchaturvedi, 2025-04-04:
Issue here: https://github.com/arduino/Arduino/issues/12041#issue-2971416201

PS I tested this on MicroPython and it works fine there FYI

iabdalkader, 2025-04-04:
PD6 is connected to the SDCARD clock. PA8 gets configured as a timer channel for the camera for the Portenta breakout.

snapchaturvedi, 2025-04-05:
> PD6 is connected to the SDCARD clock. PA8 gets configured as a timer channel for the camera for the Portenta breakout.

Are you suggesting this pin cannot be used for extint? I have been using it for quite some time now and the problem just started to come up. 

Experiencing the same issue with pins D0-D10 so I don't think it's a D6 issue (hence edited post title). 

PC

iabdalkader, 2025-04-05:
> Are you suggesting this pin cannot be used for extint

These pins get configured as output clocks by default on reset, the timer pin will output 12 or 24MHz clock so you shouldn't connect them to another output. You could reuse D6 if you don't connect the camera shield.

> Experiencing the same issue with pins D0-D10 so I don't think it's a D6 issue (hence edited post title).

I can't reproduce this, for me extint works fine. I just tested the default example that comes with the IDE

https://github.com/user-attachments/assets/57959793-e776-4de5-895f-3f9bfa4ed2f2

snapchaturvedi, 2025-04-06:
> I can't reproduce this, for me extint works fine. I just tested the default example that comes with the IDE

Thanks for testing on your end. Here's how it's going down for me:
https://drive.google.com/file/d/12uCMs4njfGJfKQpb6UPopYdc5WPCtf5f/view?usp=sharing

I just tested this without the LoRa Vision Shield and extint is working on D6 pin also. Perhaps the issue arises only when a Vision Shield is used. Could you please check if you're experiencing the same?

PC

kwagyeman, 2025-04-06:
@snapchaturvedi - I justed tested with an Arduino Portenta using the script you provided and I did not run into any issues.

I tested without any shields attached, the vision and ethernet shield attached, and the vision and lora shield attached. The script worked in each case.

snapchaturvedi, 2025-04-06:
Thanks for testing, @kwagyeman. I just tested this on a Portenta with the older Vision Shield (the one that comes with HM01B0) to confirm that I am not getting the issue on pins D0-D10. 

However, when I use Vision Shield Rev2 (the one that comes with HM0360), I am experiencing the issue. 

Could you please test on a Portenta with VS Rev2 by any chance?

PC

kwagyeman, 2025-04-06:
@snapchaturvedi - The HM0360 one works as expected for me too.

snapchaturvedi, 2025-04-06:
Thanks for checking @kwagyeman. I don't know what else to suggest. I have 4 Rev2s and am getting the same issue on all of them. I suspect all my shields are from the same production batch and hence could be a hardware issue.

All is well on my other older shields.

PC

kwagyeman, 2025-04-06:
@snapchaturvedi - Is there anything else on this issue you would like me to look at? Otherwise, I think it makes sense to close it.

iabdalkader, 2025-04-07:
> I just tested this on a Portenta with the older Vision Shield (the one that comes with HM01B0) to confirm that I am not getting the issue on pins D0-D10.
> However, when I use Vision Shield Rev2 (the one that comes with HM0360), I am experiencing the issue.

@snapchaturvedi Just to confirm, you can configure and use D0-D10 as extint wakeup with Rev1 but not Rev2? Note you still should not use D6 (or any other pin used by the shield).

As far as I know, the only update in rev2 is replacing the HM01b0 with HM0360.

closed 2025-05-14 by snapchaturvedi